GazzolaLab/PyElastica #1: Visualization

jiefengsun, 2020-05-15:
Hello,

I just want to ask what software you used to visualize the results as done in your papers.  Is it Blender?

nmnaughton, 2020-05-16:
Hi @jiefengsun, thanks for reaching out! We use POVray for our visualizations and have developed some custom scripts for the command line use of it. However, it does look like POVray can be interfaced with Blender, so that might be a good (maybe even better?) method for visualization.

jiefengsun, 2020-05-16:
Thanks for the clarification! I asked the question since I found the link to the visualization part was missing. Look forward to seeing your update on that part.

nmnaughton, 2020-05-16:
Yes, we are behind on that section of the website. We hope to have something posted soon. Thanks for your interest!

closed 2020-05-16 by nmnaughton