GazzolaLab/PyElastica #2: Applying non-constant torques and forces

pcsong33, 2020-07-02:
Hello,
Is there a way to apply non-constant forces / torques to a rod? For example, I am trying to apply torque to a rod for a certain duration of time, rather than to have the torque act on the rod for the whole duration of the simulation.

armantekinalp, 2020-07-02:
Hello,

Yes you can apply torque in a certain time interval. For that I suggest you to write your own forcing class. Below I am also adding an example code for you. 

```python
class MyNewTorqueClass(NoForces):
    def __init__(self, torque_mag: float, torque_direction, time_start: float, time_end: float):

        self.torque_mag = torque_mag 
        self.torque_direction = torque_direction # defined in global frame, shape(3,1)
        self.time_start = time_start
        self.time_end = time_end

   def apply_forces(self, system, time: np.float = 0.0):
        return

   def apply_torques(self, system, time: np.float = 0.0):
       
        if time > self.time_start and time < self.time_end:
               torque_vector = self.torque_mag * self.torque_direction 
        else:
               torque_vector = np.array([0., 0., 0.]).reshape(3,1)

        system.external_torques += _batch_matvec(
            system.director_collection, torque_vector
        )
```

There are couple of example forcing classes in [elastica/external_forces.py](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/external_forces.py), check them out as well.

pcsong33, 2020-07-06:
Thank you for your help!

closed 2020-07-06 by pcsong33