GazzolaLab/PyElastica #3: Plectoneme-Solenoid Transition

HogeLab, 2020-07-13:
Hello.

I want to simulate Plectoneme-Solenoid Transition in the paper.
How did you do it ? 
What module did you use?

armantekinalp, 2020-07-15:
Hello,

We have not included that case in our examples for PyElastica yet. For the paper it was simulated using our Cpp code. You can also set this case by yourself with PyElastica. What you have to do is:

1. Set up your client file create simulation class, rod object, append rod object to simulation etc. You can refer to one of the example cases in `examples` directory. Also I suggest you to visit our website `https://www.cosseratrods.org/software/pyelastica/` , for more explanation.

2. For Plectonme-Solenoid test case you need two boundary conditions. One boundary condition is already part of PyElastica which is `OneEndFixedRod`. You need to write second boundary condition. It should be rotating the other end of the rod through out the twist time. In order to write new boundary condition you can refer to already written boundary condition in `elastica/boundary_condition.py` . I suggest you to especially look at `HelicalBucklingBC`.

3. Finally you need to add a force at the end of the rod. For that you can use `EndPointForces` class. One example use case of this class is given in `examples/TimoshenkoBeamCase/timoshenko.py`.

Let me know if you have further questions.

HogeLab, 2020-07-15:
Thank you !　
I also want to try the simulation in the case of adding a torque at the end of the rod , but a module like `EndPointTorques` dosen't exit. 
Do I need to make that module by myself ?

armantekinalp, 2020-07-15:
Yes you need to write the class. Couple of reminders, 

1. Make sure you derived new forcing class for `EndPointTorques` from `NoForces` class. Otherwise simulation will throw an error.

2. Add your torque calculation in `apply_torque` class method.

3. We are storing external torques in material frame, so if you are applying torque in lab frame of reference then you need to multiply computed torques with `director_collection`. 

```python
class EndPointTorques(NoForces):
    def __init__(self, torque_start, torque_end):

        self.torque_start = torque_start # defined in global frame, shape(3,1)
        self.torque_end= torque_end # defined in global frame, shape(3,1)

   def apply_forces(self, system, time: np.float = 0.0):
        return

   def apply_torques(self, system, time: np.float = 0.0):

        system.external_torques[...,0] += system.director_collection[...,0] @ self.torque_start 
        system.external_torques[...,-1] += system.director_collection[...,-1] @ self.torque_end 
      
```

HogeLab, 2020-07-15:
Thank you very much.

closed 2020-07-15 by HogeLab

HogeLab, 2020-07-26:
I can't find Self-contact setting that corresponds to `useSelfContact` in elastica++. Where is it ? Will it be programmed in the future?

armantekinalp, 2020-07-26:
Yes it hasn't implemented yet. It will be implemented in future.