GazzolaLab/PyElastica #4: Visualization

adoroud, 2020-10-08:
Hi,

What is the class or commands for the octopus visualization? does Elastica (PyElastica) have an environment for the visualization part? I could not find anything in your examples providing the same outputs as in the case studies.

Thanks,

nmnaughton, 2020-10-15:
Hi @adoroud, 

We have not yet released any visualization scripts. We are preparing an update to be released soon and hope to include some there.

adoroud, 2020-10-15:
> Hi @adoroud,
> 
> We have not yet released any visualization scripts. We are preparing an update to be released soon and hope to include some there.

Thank you for your reply,
Azadeh

nmnaughton, 2020-11-02:
Hi @adoroud 

We just released a new version of PyElastica with an example script for how we do our visualizations. It is located [here]
(https://github.com/GazzolaLab/PyElastica/tree/master/examples/Visualization). Hope this is helpful!

closed 2020-11-02 by nmnaughton

adoroud, 2020-11-02:
Hi,

Thanks a lot for informing me.

Best,
Azadeh