GazzolaLab/PyElastica #7: Applying non-constant torques in one direction

ing1992, 2020-11-30:
Hello!
Is there any way to apply torque( at the end of the rod) in the same direction as rod is fixed. For example I have a rod in y direction and I would like to apply a torque in this direction. I have created a calss called "OnedirectionTorque"

class OnedirectionTorque(NoForces):
    def __init__(self,torque_end,direction):
        super(EndPointTorques, self).__init__()
        self.torque_end= (torque_end * direction).reshape(3, 1) # defined in global frame, shape(3,1)
    def apply_torques(self, system, time: np.float = 0.0):
        system.external_torques[...,  -1] += system.director_collection[..., -1],self@torque_end 

I also have add it this piece of lines in examples/TimoshenkoBeamCase/timoshenko.py
direction = np.array([0.0, 1.0, 0.0])
toque_end = -10
timoshenko_sim.add_forcing_to(shearable_rod).using(
    UniformTorques,toque_end,sdirection)

Is it implemented correctly

Thanks in advance

armantekinalp, 2020-11-30:
Hello @ing1992 

Your `OnedirectionTorque` implementation looks right. This class will apply a torque on the last element of the rod in the direction set by `direction` variable. 

However, I think you should change the lines of code you added in your `timoshenko.py` file. 
This code is using the `UniformTorques` class which is some other class to generate uniform torques.
`timoshenko_sim.add_forcing_to(shearable_rod).using(
UniformTorques,toque_end,direction)`

I suggest you to do the following change to use `OnedirectionTorque`  class.
`timoshenko_sim.add_forcing_to(shearable_rod).using(
OnedirectionTorque,toque_end,direction)`

Let me know if this helps.

ing1992, 2020-11-30:
Thank you for answering my question
I have tried your suggestion (but forgot to change it whene I wrote the
message)
Code runs correctly but does not produce any change on the rod!

armantekinalp, 2020-11-30:
Since the torque you are applying is in the rod direction, it generates torsion on the rod. You can check if your code works or not by just printing the twist value of the rod, which can be accessed by `shareable_rod.kappa[2]` . I suggest printing out the value `kappa[2]` and see if it is changing or not. 

If you are interested in rod buckling, you should disturb the rod. Otherwise rod won't change its shape even though it is twisting. I suggest to add a small velocity disturbance at the middle node or nodes of the rod in `normal` or `binormal` direction, after you initialize the rod. You can play around with the value of the disturbance.

`sherable_rod.velocity_collection[..., int(n_elem/2)][0] += 1E-3`

You should see some movements in the rod, like it is buckling. However, if you want to generate plectonemes or solenoids, rod self-contact should be used. But we haven't released the code for self-contact yet, and planning to release it in future.

Let me know this helps.

ing1992, 2020-12-01:
Thank you @armantekinalp

closed 2020-12-01 by ing1992

armantekinalp, 2021-07-29:
Hi @Seren1ty-bot ,

I think some variable is missing when you do `add_forcing_to`. Could you please check the variable names and make sure you add everything that `UniformTorques` class uses to during initialization step?

If that not works please share the part of the code, where you add `UniformTorques` class in to your simulation.

armantekinalp, 2021-07-31:
Hi @Seren1ty-bot  

Couple of things:

1) Are there any other forcing class added to your simulation?

2) Have you changed anything in `UniformTorques` class? Can you try this

`timoshenko_sim.add_forcing_to(shearable_rod).using(
    UniformTorques,
    torque  = torque,
    direction  = direction1
)`

3) You don't need to add `OnedirectionTorque` class in to the core PyElastica files. You can have it on another script and import it from there. Keep in mind that you should import `NoForces` class like this `from elastica.external_forces import NoForces`. 

Error you are getting, can be due to two reasons. First one is a variable is missing (or name typo) when you add forcing class to the simulation. Second one is during the forcing class initialization there is an error in the code.  Maybe first try my suggestions above and if those  do not work share the script here.

armantekinalp, 2021-07-31:
These are the problems I see 

1. In your `untitled0.py` file you import `endpointtorque` but the name of the class is `EndPointTorque`.
2.  In your `endpointtorque.py`  you have some mistakes. First you cannot write `def init`, you should write `def __init__(self, ....) `
2. Second `apply_torques` method, you should first multiply end point torque with directors and then add to the external_torques . 

here is my corrections

``` python
class EndPointTorque(NoForces):
    def __init__(self,torque,direction):
        self.torque= (torque * direction).reshape(3, 1) # defined in global frame, shape(3,1)
    def apply_torques(self, system, time: np.float = 0.0):
        system.external_torques[..., -1] += system.director_collection[..., -1] @ self.torque
```

armantekinalp, 2021-08-03:
Hi @Seren1ty-bot 

Below code is running on my local device. Can you copy on your local and try it again.

``` python

import numpy as np
# Import Wrappers 
from elastica.wrappers import BaseSystemCollection, Connections, Constraints, Forcing, CallBacks
# Import Cosserat Rod Class
from elastica.rod.cosserat_rod import CosseratRod
# Import Boundary Condition Classes
from elastica.boundary_conditions import OneEndFixedRod, FreeRod
from elastica.external_forces import NoForces, EndpointForces, UniformTorques, UniformForces
# Import Timestepping Functions
from elastica.timestepper.symplectic_steppers import PositionVerlet
from elastica.timestepper import integrate
from elastica.callback_functions import CallBackBaseClass
from collections import defaultdict
# from elastica import endpointtorque

class TimoshenkoBeamSimulator(BaseSystemCollection, Connections, Constraints, Forcing, CallBacks):
    pass

timoshenko_sim = TimoshenkoBeamSimulator()

# setting up test params
n_elem = 100

density = 1000
nu = 0.1
E = 1e6
# For shear modulus of 1e4, nu is 99!
poisson_ratio = 99

start = np.zeros((3,))
direction = np.array([0.0, 0.0, 1.0])
normal = np.array([0.0, 1.0, 0.0])
base_length = 3.0
base_radius = 0.25
base_area = np.pi * base_radius ** 2

shearable_rod = CosseratRod.straight_rod(
    n_elem,
    start,
    direction,
    normal,
    base_length,
    base_radius,
    density,
    nu,
    E,
    poisson_ratio,
)

timoshenko_sim.append(shearable_rod)

timoshenko_sim.constrain(shearable_rod).using(
    OneEndFixedRod, 
    constrained_position_idx=(0,), 
    constrained_director_idx=(0,)
)
print('One end of the rod is now fixed in place')

origin_force = np.array([0.0, 0.0, 0.0])
end_force = np.array([-10.0, 0.0, 0.0])
ramp_up_time = 5.0

timoshenko_sim.add_forcing_to(shearable_rod).using(
    EndpointForces, 
[... 127 lines cut ...]
    metadata = dict(title="Movie Test", artist="Matplotlib", comment="Movie support!")
    writer = FFMpegWriter(fps=fps, metadata=metadata)
    fig = plt.figure()
    plt.axis("equal")
    with writer.saving(fig, video_name, dpi=100):
        for time in range(1, len(plot_params["time"])):
            
            x = positions_over_time[time][2]
         
            y = positions_over_time[time][0]
            fig.clf()
            plt.plot(x, y, "-", linewidth=3)
            
            plt.xlim([0 - margin, 3 + margin])
            plt.ylim([-0.05 - margin, 0.01 + margin])
            writer.grab_frame()
    plt.close(fig)

filename_video = "continuum100.mp4"
plot_video(pp_list, video_name=filename_video, margin=0.02, fps=25)
    
Video("continuum100.mp4")

print(shearable_rod.kappa[2])
```