GazzolaLab/PyElastica #8: Errors with the notebooks

rohitkjohnedu, 2020-12-30:
Hello,

I was running the code in the notebook for the Timoshenko beam (https://hub.gke2.mybinder.org/user/gazzolalab-pyelastica-0ttn23rh/notebooks/examples/Binder/1_Timoshenko_Beam.ipynb). I got an error in the last notebook cell, and it said the following "AttributeError: 'PositionVerlet' object has no attribute 'do_step'". How do I fix this error?

In the slithering snake tutorial, I got an error saying "TypeError: __init__() missing 1 required positional argument: 'rest_lengths'". I found there is a mistake in the code in the cell under the heading "Muscle Torques", there should be an additional argument to the function snake_sim.add_forcing_to(shearable_rod).using(). It is missing the argument 'rest_lengths=shearable_rod.rest_lengths'. Adding this argument fixed it. I would recommend updating the code.

Kind regards,
Rohit K. John

armantekinalp, 2020-12-31:
Hello Rohit,

Thanks for pointing out the errors. I updated both Jupyter Notebooks  for both `1_Timoshenko_Beam.ipynb` and `2_Slithering_Snake.ipynb` , now both of them work fine.  If you are running notebooks on the Binder there is still a bug for making the slithering snake videos, but you should be able to run it on your local with `FFMPEG`. 
Please let me know if you see any other problems?

Best,

Arman

nmnaughton, 2021-01-19:
When we updated the new version of PyElastica, we accidentally deleted the `apt.txt` file from the main directory which tells Binder to install ffmpeg. I readded the file (and also readded the `continuum_snake.mp4` to the assets folder, which also got overwritten). Ffmpeg works on Binder now.

closed 2021-01-19 by nmnaughton