GazzolaLab/PyElastica #9: Path Error

alantes, 2021-03-01:
The examples/Visualization/ContinuumSnakeVisualisation/continuum_snake_render.py file cannot run correctly. The error is: ModuleNotFoundError: No module named 'examples'.
I think it may be be path error.

armantekinalp, 2021-03-01:
@alantes Thanks for pointing out this issue.  PR #10  should solve this problem.

alantes, 2021-03-02:
> @alantes Thanks for pointing out this issue. PR #10 should solve this problem.

Thank you for your marvelous work! But I have to remind that there are some other files share the same issue, too (e.g. helicalbuckling.py). It would be particularly nice if you can reexamine the files.

armantekinalp, 2021-03-02:
@alantes Are you using the current master branch, or did you ` pip install PyElastica`? Also can you share the error message here. Thank you.

alantes, 2021-03-02:
I am using the current master branch. Downloaded the source code and run in the terminal
python continuum_snake.py
python continuum_snake_render.py
The first one is okay but the second one is followed by:
from examples.Visualization._povmacros import Stages, pyelastica_rod, render
ModuleNotFoundError: No module named 'examples'
Sorry for my unfamiliarity with coding, otherwise I can debug it myself.

armantekinalp, 2021-03-02:
Okay can you clone again from the master branch and run `python continuum_snake_render.py`. Path for `examples` directory was added with our last PR.

alantes, 2021-03-02:
Thanks a lot for your patience!! It's now working.

armantekinalp, 2021-03-02:
You are welcome. I am closing this issue if you have any more question please feel free to open this issue or create a new one.

closed 2021-03-02 by armantekinalp