GazzolaLab/PyElastica #11: paralleled rod connection setting

daphne198693, 2021-03-29:
Dear PyElastica,

thank you in advance. 
While using powerful PyElastica to model a soft robot, which is consisted of 3 parallel rods connected from each other. Luckily, I found a similar one from your paper: Topology, Geometry, and Mechanics of Strongly Stretched and Twisted Filaments: Solenoids, Plectonemes, and Artificial Muscle Fibers, but i can not find any reference code.
So could you share the modeling code like boundary condition or how the bilateral rods connected to the centeral one.
Thank you for your time and help.

BR
Daphne

daphne198693, 2021-03-29:
i mean the parallel rod in which the end point directions are the same, like they are bonded.

armantekinalp, 2021-03-30:
Hi Daphne,

Thanks for reaching out. Could you please more specific which example (Fig?) are you referring in `Topology, Geometry, and Mechanics of Strongly Stretched and Twisted Filaments: Solenoids, Plectonemes, and Artificial Muscle Fibers` paper.

daphne198693, 2021-03-30:
Hi Armantekinalp,

thanks a lot for your prompt reply. 
yea, exactly this paper. but i'm not doing HelicalBuckling simulation, just three rods attached together, once one of them gets stain and elongated, rest of them would be elongated proportionally. 
i have drawn a draft.  
[image]
Thanks a lot for your helping.

armantekinalp, 2021-03-31:
Hi @daphne198693,

First, in PRL paper we applied two end boundary condition on the rod. So it is not a relevant to the problem you are trying to solve. 

Second, for your experiment you should constrain relative position and orientation of the rods. 
For example let's take two rods. You want to constrain distance between the surfaces of these two rods. In other words you should apply a restoring force on both rods like 

F = k (x1 - x2) 

where k is spring constant, x1 and x2 are the surface positions of rod1 and rod2 respectively. Next step is to constrain their relative orientation. Similar to constraining relative position, you should apply restoring torques now. I suggest you to look at the examples in `examples/JointCases/` , especially `fixed_joint.py` and the connection functions we used.

Currently we are working on a a general framework that can help your problem. I will let you know when we release it.

Let me know if you have any other questions.

daphne198693, 2021-04-01:
Hi Armantekinalp,

Thanks very much for your so helpful suggestion and so detailed guide.
That's amazing news that you will release more general frameword, hope it will not long. :)
now I'm exactly modeling start with using one rod whose two ends with fixed joints to make constrain with others rods, and try to restore torques with UniformTorques. If there are some example of adding different types external force that will very useful.
Again, thank you a lot for your time and help.

daphne198693, 2021-04-01:
BTW,  could Pyelastica support polymer foam this kind of material simulation, can be compressed or elongated with high percentage? 
like Young's Elastic Modulus range is from how much? 
i have tried quite a lot combinations, seems not get good result. if you could give some suggestion it will be grateful :)

armantekinalp, 2021-04-01:
> Hi Armantekinalp,
>
> Thanks very much for your so helpful suggestion and so detailed guide.
> That's amazing news that you will release more general frameword, hope it will not long. :)
> now I'm exactly [...]

So you want to be careful here. `fixed_point.py` is an example to join two rods not by side by side but at the rod ends. You can see that if you run the scripts and check the videos. In your case, you want to attach two rods side by side so you need to write your own `JointClass` for that. However the high-level idea is the same, i.e. you fix relative position and orientation of rods by applying forces and torques.  

'UniformTorques' class applies user defined torque vector on all elements of the rod. I don't think you can use it for constraining relative orientation of two rods. Instead I suggest writing  your own `JointClass`.

armantekinalp, 2021-04-01:
> BTW, could Pyelastica support polymer foam this kind of material simulation, can be compressed or elongated with high percentage?
> like Young's Elastic Modulus range is from how much?
> i have tried [...]

By default PyElastica uses a linear elastic model and rods are treated as incompressible. Do you have a constitutive model for your polymer foam?  Is it linear elastic?

daphne198693, 2021-04-01:
> By default PyElastica uses a linear elastic model and rods are treated as incompressible. Do you have a constitutive model for your polymer foam? Is it linear elastic?

Thanks for your prompt reply. :)
Yea, it's linear elastic. Actually, we focus on modeling some kind of soft robot, and i do not have a constitutive model for the polymer foam yet, but whose character is similar to spring.

daphne198693, 2021-04-05:
Hi Armantekinalp,
as you mentioned Pyelastica is incompressible, but i tried different combinations of rob parameters, i did not find the way to make the single rod elongate. do you have any suggestion?
thank you a lot

armantekinalp, 2021-04-05:
Hi @daphne198693 

Can you be more specific? For example what are you boundary conditions, external forces, rod parameters?

daphne198693, 2021-04-14:
Thank you @armantekinalp for your so patient and helpful support. 
we are keeping attention on your latest release. 
thank you very much

closed 2022-06-02 by armantekinalp