GazzolaLab/PyElastica #13: Static Analysis of the Rod

emersonkt, 2021-04-06:
Hello PyElastica,

In my project, I do a static analysis of 3 rods in parallel. So far, two examples that I have seen are that of timoshenko and the snake, but both are dynamic models.

To do this, I thought of setting the force ramp time close to zero compared to the final time and "_nu_" large enough to "dissipate" the energy and converge as fast as possible. Despite reaching the desired final configuration, the simulation is still dynamic and I would like to only get the configuration for a given force, for example. Is there a way to make a static model?

I hope I have been clear enough and thank you in advance for all the technical support.

armantekinalp, 2021-04-06:
Hi @emersonkt,

> In my project, I do a static analysis of 3 rods in parallel. So far, two examples that I have seen are that of timoshenko and the snake, but both are dynamic models.

Do you have 3 connected rods in your simulations or 1? Connection of  3 rods were asked in another issue #11. Maybe read the discussion there and let me know if you have any questions.

> To do this, I thought of setting the force ramp time close to zero compared to the final time and "_nu_" large enough to "dissipate" the energy and converge as fast as possible. 
 
There might be an issue in your approach. You need to ramp the force in a finite amount of time otherwise numerical instabilities can happen. This finite time depends on the load and rod properties. You should determine this minimum ramp time by observing the dynamic simulation. In order to see if there is some numerical instability.  Best way to do that is plotting the motion of the rod and checking if there are any `Nan` in `rod.position_collection`, `rod.velocity_collection`, etc.  
Also as you said, increasing `nu` will damp out the transient and you will reach the static configuration faster.

> Despite reaching the desired final configuration, the simulation is still dynamic and I would like to only get the configuration for a given force, for example. Is there a way to make a static model?

Finally in PyElastica we do not have an option make a static model or compute static solution directly without doing time integration.

emersonkt, 2021-04-07:
Thank you very much for your prompt feedback and your observations. It has helped me to clarify on the issue.

closed 2021-04-07 by emersonkt