GazzolaLab/PyElastica #15: how to load a custom model and apply this rod algorithm

xiaoerlaigeid, 2021-04-09:
Very impressive Jobs! I am attempt to use this library in vascular intervention simulation platform. I want to load vascular model  and apply this rod as catheter. But I don't know how to load my model(a STL file)  and  make the rod have collision and friction. Could you give me some demo ? Thanks!

armantekinalp, 2021-04-11:
Hi @xiaoerlaigeid,

Currently we don't have a support to generate a rod from a STL file. However, to generate a rod you need `position` of nodes and `radius` of elements. I think If you have an STL file you can generate those.  Regarding the contact(collision) model PyElastica only have contact model between rod and rigid cylinder, but in our future we will have rod-rod and self contact. Finally, we have a friction model between surface of a plane and a rod in PyElastica. Will this type of friction model be useful for you? Let me know if you have any questions.

xiaoerlaigeid, 2021-04-12:
Thank you for your reply.  I mean I want to load an STL model which is used for the environment.   Or just using some cube and sphere tossss replace STL environment.   Do you have a demo like this  ?

armantekinalp, 2021-04-12:
We are planning to develop an STL support in future but unfortunately we don't have a STL support and demo right now.

skim0119, 2022-08-02:
We are not currently planning to include this feature. We'll reopen if this is needed in the future.

closed 2022-08-02 by skim0119