GazzolaLab/PyElastica #16: Issues defining compute_and_plot_velocity(plot_params: dict, period) function

ilanataha, 2021-04-12:
I've been trying to write a plotting function for different rod systems just to get a feel for this software. I have based my code on the `compute_and_plot_velocity` function defined in the Slithering Snake tutorial. I have also  imported all the wrappers mentioned. 

However I run into the same error every time: "from analysis_functions import compute_projected_velocity
ModuleNotFoundError: No module named 'analysis_functions'". I tried looking for analysis_functions through documentation but to no avail. What am I missing?

The definition of the function here attached:

```
    def compute_and_plot_velocity(plot_params: dict, period):
        from matplotlib import pyplot as plt
        from analysis_functions import compute_projected_velocity

        time_per_period = np.array(plot_params["time"]) / period
        avg_velocity = np.array(plot_params["avg_velocity"])

        [velocity_in_direction_of_rod,
         velocity_in_rod_roll_dir,
         avg_forward,
         avg_lateral] = compute_projected_velocity(plot_params, period)

        print("average forward velocity:", avg_forward)
        print("average forward lateral:", avg_lateral)

        fig = plt.figure(figsize=(5, 4), frameon=True, dpi=150)
        ax = fig.add_subplot(111)
        ax.grid(b=True, which="major", color="grey", linestyle="-", linewidth=0.25)
        ax.plot(time_per_period[:], velocity_in_direction_of_rod[:, 2], "r-", label="forward")
        ax.plot(time_per_period[:], velocity_in_rod_roll_dir[:, 0], "b-", label="lateral", )
        ax.plot(time_per_period[:], avg_velocity[:, 1], "g-", label="normal")
        ax.legend(prop={"size": 12})
        ax.set_ylabel('Velocity (m/s)', fontsize=12)
        ax.set_xlabel('Time (s)', fontsize=12)
        plt.show()

```

Any help would be of great assistance.
Thank you

armantekinalp, 2021-04-12:
Hi @ilanataha,

Are you running this on your local system or on Binder? It seems to me that python cannot find and import from `analysis_functions.py` . So can you check if the script you are wrote ` compute_and_plot_velocity` function and `analysis_functions.py` are in the same directory?

ilanataha, 2021-04-13:
I tried moving the script into the pyelastica folder of the project, it didn't help. But I checked and it seems that` analysis_functions.py` is nowhere to be found in the folder itself, or anywhere on my computer, really. I'll maybe try reinstalling the package.

armantekinalp, 2021-04-13:
If you install the package with `pip install pyelastica`, `examples` folder won't be installed. Only PyElastica source code will be installed. Try to clone or download PyElastica from GitHub to your local machine. Then copy/move the `analysis_functions.py` in your working directory.

closed 2022-01-08 by armantekinalp