GazzolaLab/PyElastica #17: Simulating Soft Arm in Pyelastica

Abdelrahman-Alkhodary, 2021-07-09:
Hello all,

I have a task in my project to simulate a soft robotics arm that is actuated by 12 cables connected to 3 discs with four cables connected to each disc. The three discs are distributed over the arm with one close to the base, one in the middle and the last close to the end of the arm. I would like to ask if it's possible to use pyelastica to simulate the arm as while I was going through the website, I could not find a way to attach any cables to the arm, If so, can you please give me some advice how to simulate it.

nmnaughton, 2021-07-31:
If you want to attach cables you will need to treat each cable as an individual rod that you attach to the main arm using joints (https://www.cosseratrods.org/cosserat_rods/multiple-rods/). To model the disks you likely need to create custom boundary conditions that model the effects of the disks as there are not currently any routines written that do this.

skim0119, 2022-01-23:
I’m closing this issue because it has been inactive for more than 6 months.

@Abdelrahman-Alkhodary You are welcome to reopen and contact us if you would like to try out PyElastica.

Thank you!

closed 2022-01-23 by skim0119