GazzolaLab/PyElastica #21: Point Force Class

michael-baier, 2021-09-18:
Hello, 
Is there a way to apply a force to the Rod (for example in the middle)? how can i implement it? 

Any help would be of great assistance.
Thank you

armantekinalp, 2021-09-18:
Hi @michael-baier ,

Yes. PyElastica allows you to apply forces on rods. You can use the default `forcing` classes available in PyElastica or you can write on your own. In order to see available forcing classes and their implementation please check this [link](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/_elastica_numpy/_external_forces.py)

However, if you want to apply a force to the  middle node of a rod, you should write your own forcing class. Here is an example,

```python

class PointForceOnMiddleNode(NoForces):
    def __init__(self, point_force = np.array([0, 0, 0])):
        super(PointForceOnMiddleNode, self).__init__()
        self.point_force = point_force.reshape(3,1)

    def apply_forces(self, system, time=0.0):
        n_nodes = system.n_elems + 1
        system.external_forces[:,int(n_nodes/2)] += self.point_force 

```

michael-baier, 2021-09-19:
Thank you very much! You helped me to solve my problem.

closed 2021-09-19 by armantekinalp