GazzolaLab/PyElastica #28: Timoshenko example throws an error on the current `master`

tp5uiuc, 2021-12-24:
Running the Timoshenko example, from `examples/TimoshenkoBeamCase/timoshenko.py` throws the following error.

```
Traceback (most recent call last):
  File "/Users/tp5/code/PyElasticaPublic/examples/TimoshenkoBeamCase/timoshenko.py", line 41, in <module>
    shearable_rod = CosseratRod.straight_rod(
TypeError: straight_rod() missing 1 required positional argument: 'poisson_ratio'
```

tp5uiuc, 2021-12-24:
I'm on the current `master` branch.

If it helps, here's my list of installed packages. Some of them may not be relevant. 

```
Package                   Version
------------------------- -----------
aiohttp                   3.7.4.post0
anyio                     2.1.0
appdirs                   1.4.4
appnope                   0.1.0
argon2-cffi               20.1.0
args                      0.1.0
astor                     0.8.1
astroid                   2.4.2
async-generator           1.10
async-timeout             3.0.1
attrs                     20.3.0
autobahn                  21.3.1
Automat                   20.2.0
Babel                     2.9.0
backcall                  0.2.0
bashate                   2.0.0
black                     20.8b1
bleach                    3.2.1
cachelib                  0.1
certifi                   2020.12.5
cffi                      1.14.4
chardet                   3.0.4
click                     7.1.2
clint                     0.5.1
cmakelint                 1.4.1
colorama                  0.4.4
constantly                15.1.0
CppHeaderParser           2.7.4
cryptography              3.4.8
cssselect                 1.1.0
cycler                    0.10.0
decorator                 4.4.2
defusedxml                0.6.0
Deprecated                1.2.11
dill                      0.3.3
docopt                    0.6.2
entrypoints               0.3
flake8                    3.8.4
Flask                     2.0.1
future                    0.18.2
h5py                      3.5.0
howdoi                    2.0.10
hyperlink                 21.0.0
idna                      2.10
imageio                   2.9.0
incremental               21.3.0
iniconfig                 1.1.1
ipykernel                 5.3.4
ipython                   7.19.0
ipython-genutils          0.2.0
ipywidgets                7.5.1
isort                     5.6.4
itsdangerous              2.0.1
jedi                      0.17.2
Jinja2                    3.0.1
joblib                    1.0.1
json-rpc                  1.13.0
json5                     0.9.5
jsonschema                3.2.0
jupyter                   1.0.0
jupyter-client            6.1.7
jupyter-console           6.2.0
jupyter-core              4.7.0
[... 98 lines cut ...]
threadpoolctl             2.1.0
tifffile                  2021.7.30
toml                      0.10.2
tornado                   6.1
tqdm                      4.54.0
traitlets                 5.0.5
Twisted                   21.7.0
txaio                     21.2.1
typed-ast                 1.4.1
typing-extensions         3.7.4.3
urllib3                   1.26.2
vtk                       9.0.3
wcwidth                   0.2.5
webencodings              0.5.1
websockets                8.1
Werkzeug                  2.0.1
westerly                  0.2.1
wheel                     0.36.2
widgetsnbextension        3.5.1
wrapt                     1.12.1
wslink                    1.0.7
yarl                      1.6.3
zope.interface            5.4.0
```
On my side, let me see if a fresh download solves this issue.

tp5uiuc, 2021-12-24:
Update: Even a fresh download of the current master of pyelastica doesn't solve the issue.

armantekinalp, 2021-12-24:
Please install version 0.1.0.post5

tp5uiuc, 2021-12-24:
The issue is fixed, thanks @armantekinalp for pointing it out.

I have `pyelastica-post3` installed system-wide on my computer via `pip`. But the current master version I downloaded from `git` to run the examples is `pyelastica-post5`. For reference, the example files are only accessible in the git repository. 
But when I run the examples (even within the git repository), an `import pyelastica` statement first picks up the system-wide installed `pyelastica-post3` and not `pyelastica-post5` (even if its added to the path). Updating my system-wide installation to ``pyelastica-post5` resolved this issue.

closed 2021-12-24 by tp5uiuc