GazzolaLab/PyElastica #29: Is there a cpp version of this library?

xarthurx, 2021-12-30:
Hi, I'm developing a research project and found this library.
However, I'm doing mainly cpp and need a lib to be able to integrate it into the current system.

After checking the code, it seems the main lib is written in python?
Or do you have another version that is written in C/C++ elsewhere?

Thank you

tp5uiuc, 2021-12-30:
Hey @xarthurx, thanks for the question.

We are currently developing a C++ version of Elastica, but it's not been released to the public yet. We expect a release sometime soon, in a couple of months. 

Meanwhile, if you really need a code that simulates Cosserat rods, please have a look at the (older) version of [Elastica here](https://github.com/mattialab/elastica). It's written in C++ and maybe what you are looking for.

xarthurx, 2022-01-04:
@tp5uiuc  Thank you for the info.
I checked the original version (non-HPC). It seems it is only tested on Mac and Linux.

Though I can make it work in Windows, I would suggest that you provide the library in CMake system that can be easily imported as a cross-platform library.

tp5uiuc, 2022-01-04:
Thanks for the suggestion @xarthurx, we will keep that in mind. I am closing the issue, feel free to reopen if you have more questions.

closed 2022-01-04 by tp5uiuc

xarthurx, 2022-01-17:
@tp5uiuc  I'm currently developing a project that would need a simulator like this.
Would it be possible to have early access to the new version and contribute to it as well?
I've developed several projects for published CG papers, and am quite familiar with cpp.

(Asking because I don't want to integrate the old one just a few months before your new release...)

tp5uiuc, 2022-01-18:
@xarthurx Let me get back to you on that. I would be happy to share it immediately if I was the only stakeholder. But I need to consult with my advisor first. Hope you understand :)

xarthurx, 2022-02-02:
@tp5uiuc  Any update on this?

tp5uiuc, 2022-02-02:
Hey @xarthurx, thanks for waiting. We are finishing up the package now (it's not entirely done), and I estimate it to take some time between 5-7 weeks. My advisor first wants to see a pre-print from our side before releasing the package, so I don't have full autonomy on the package release. So, if possible, can you please wait for the time mentioned? Thanks for understanding :)

xarthurx, 2022-02-02:
OK, then I may start to use the old one first…
Thanks for the update.

IasonManolas, 2022-07-05:
Hi are there any updates regarding the c++ release of the library?

skim0119, 2022-07-06:
@IasonManolas We are planning to release within 1~2 month.