GazzolaLab/PyElastica #30: New release (v0.2.0) breaks on import

nmnaughton, 2022-01-03:
I updated to the new v0.2.0 and am unable to import elastica. v0.1.0.post5 and before work fine. I have verified this occurs on both mac and linux. It also occurs if I directly use the main branch. Shouldn't CI have caught this? 

```
mechse-mgazzola-grp:docs naughton$ python3
Python 3.8.5 (default, Jul 21 2020, 10:48:26) 
[Clang 11.0.3 (clang-1103.0.32.62)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import elastica
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/elastica/__init__.py", line 5, in <module>
    from elastica.wrappers import *
  File "/usr/local/lib/python3.8/site-packages/elastica/wrappers/__init__.py", line 8, in <module>
    from .base_system import BaseSystemCollection
  File "/usr/local/lib/python3.8/site-packages/elastica/wrappers/base_system.py", line 13, in <module>
    from elastica.wrappers.memory_block import construct_memory_block_structures
  File "/usr/local/lib/python3.8/site-packages/elastica/wrappers/memory_block.py", line 8, in <module>
    from elastica.memory_block import MemoryBlockCosseratRod, MemoryBlockRigidBody
  File "/usr/local/lib/python3.8/site-packages/elastica/memory_block/__init__.py", line 8, in <module>
    from elastica.reset_functions_for_block_structure._reset_ghost_vector_or_scalar import (
ModuleNotFoundError: No module named 'elastica.reset_functions_for_block_structure'
>>> 
```

skim0119, 2022-01-03:
@nmnaughton Hi -- I cannot reproduce the error.

- Did you install pyelastica using pip? If yes, could you double-check `pip list` that if the new version is installed correctly? Also, check if python and pip belong the same directory.
  - I'm assuming the install is done properly, since 'elastica.memory_block' directory is added in v0.2.
- Are you running the code inside the repository clone? If yes, `from elastica...` in the last error message might be linking the old version of the code.
  - I'm thinking this is probably what is going on. Please check if you have any `sys.path.append(...)` that includes one of elastica clone.

I tried to replicate the error by
- Creating new conda environment with `python==3.8.5`
- Install pyelastica using `pip install pyelastica`
- Run `python3` and `import elastica`

```bash
(test) skim0119@wirelessprv-10-194-10-240 [00:50:19] [~/github/elastica-main] [master *]
-> % which pip; which python; python3 --version
/Users/skim0119/miniconda3/envs/test/bin/pip
/Users/skim0119/miniconda3/envs/test/bin/python
Python 3.8.5
(test) skim0119@wirelessprv-10-194-10-240 [00:50:26] [~/github/elastica-main] [master *]
-> % pip list
Package         Version
--------------- ---------
certifi         2021.10.8
cycler          0.11.0
fonttools       4.28.5
kiwisolver      1.3.2
llvmlite        0.34.0
matplotlib      3.5.1
numba           0.51.0
numpy           1.22.0
packaging       21.3
Pillow          9.0.0
pip             21.2.4
pyelastica      0.2.0
pyparsing       3.0.6
python-dateutil 2.8.2
scipy           1.7.3
setuptools      58.0.4
six             1.16.0
tqdm            4.62.3
wheel           0.37.0
(test) skim0119@wirelessprv-10-194-10-240 [00:50:54] [~/github/elastica-main] [master *]
-> % python3
Python 3.8.5 (default, Sep  4 2020, 02:22:02)
[Clang 10.0.0 ] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import elastica
>>> elastica.rod
<module 'elastica.rod' from '/Users/skim0119/github/elastica-main/elastica/rod/__init__.py'>
>>> elastica.reset_functions_for_block_structure
<module 'elastica.reset_functions_for_block_structure' (namespace)>
>>> exit()
(test) skim0119@wirelessprv-10-194-10-240 [00:51:23] [~/github/elastica-main] [master *]
-> %
```

armantekinalp, 2022-01-03:
Hi @skim0119,

I was able to generate the error. Here are the list of the packages:

```bash
(pyelastica_test_env) [atekinal@login01 atekinal]$ pip list
Package         Version
--------------- ---------
certifi         2021.10.8
click           6.7
configobj       5.0.6
cryptography    2.5
cycler          0.11.0
fonttools       4.28.5
globus-cli      1.10.1
globus-sdk      1.7.1
jmespath        0.9.2
kiwisolver      1.3.2
llvmlite        0.34.0
matplotlib      3.5.1
numba           0.51.0
numpy           1.22.0
packaging       21.3
Pillow          9.0.0
pip             21.2.4
pyelastica      0.2.0
PyJWT           1.7.1
pymp-pypi       0.4.5
pyparsing       3.0.6
python-dateutil 2.8.2
scipy           1.7.3
setuptools      58.0.4
six             1.16.0
tqdm            4.62.3
wheel           0.37.0
```

I can only generate the error, If I am not inside a directory which `elastica` is not present as a directory.

skim0119, 2022-01-03:
Okay. The issue is in `pypi`: the zip-file uploaded in `pypi` does not include `reset_functions_for_block_structure` directory. @armantekinalp Could you try to re-package and upload to `pypi`?

skim0119, 2022-01-03:
Its probably because we don't have `__init__.py` in that module. Without that file, It is only registered as namespace.

armantekinalp, 2022-01-03:
Hi @nmnaughton ,

Release `0.2.0.post1` fixes the issue, please `pip install pyelastica` . 

I am closing this issue.

closed 2022-01-03 by armantekinalp