GazzolaLab/PyElastica #32: Creating container (Docker, etc) for elastica

skim0119, 2022-01-06:
[image] **Issue by [nmnaughton](https://github.com/nmnaughton)**
_Friday Mar 20, 2020 at 16:53 GMT_
_Originally opened as https://github.com/GazzolaLab/elastica-python/issues/64_

----

We should look into creating containers for pyelastica. This will allow pyelastica to be run on clusters where users do not have permissions to install packages such a numba. 

Options are:
Docker - https://docs.docker.com/
Spack - https://spack.readthedocs.io/en/latest/containers.html
Singularity - https://sylabs.io/docs/

skim0119, 2022-01-06:
[image] **Comment by [nmnaughton](https://github.com/nmnaughton)**
_Saturday Oct 03, 2020 at 00:18 GMT_

----

 @kshivvy @armantekinalp 

Should we revisit this in light of Keshav's progress? It didn't work on Blue Waters, but his Docker file might be a good template to get started on this.

bhosale2, 2022-08-01:
@mstoelzle I realized that you may have some experience in this context (a brief look into your group repositories showed revealed docker images of elastica-related packages). Do you wish to maybe take a stab at this issue?

mstoelzle, 2022-08-04:
> @mstoelzle I realized that you may have some experience in this context (a brief look into your group repositories revealed docker images of elastica-related packages). Do you wish to maybe take a stab at this issue?

@bhosale2 Thanks for your message. Yes, I have some experience with Docker containers, but not with Singularity containers. 

To me, the purpose and motivation for this feature is not clearly defined enough yet. In most cases, the package is easy enough to install. Have you confirmed that 'numba' cannot be installed without admin rights even with the pip --user flag?

- If the purpose is to run long simulations on a cluster without any additional (complicated) dependencies, would there maybe a more basic / simple solution around to install numba?
- If the purpose is to do RL / machine learning on a cluster, I would rather want to build on a PyTorch / TensorFlow / Cuda Docker images, as it is pretty straight forward to install PyElastica on these containers instead of the alternative of adding Cuda / PyTorch to an existing, custom PyElastica image, which is a pain. 
- Are there any other applications invisioned for this container? For example an interplay with ROS? Then it would make more sense to build on a ROS image, which we are doing in our lab.

armantekinalp, 2022-08-06:
At this point we package using `pypi`. In future if we replace backhand code with `c++` or `cython` then we can reopen this issue.

closed 2022-08-06 by armantekinalp