GazzolaLab/PyElastica #43: Install failure on Windows

xarthurx, 2022-01-17:
**Describe the bug**
When install the package on Windows, the installation failed by providing the info:
```
 ERROR: Command errored out with exit status 1:
     command: 'c:\users\xarthur\source\tmprepo\pyelastica\scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xarthur\\AppData\\Local\\Temp\\pip-install-c20v04d_\\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\\setup.py'"'"'; __file__='"'"'C:\\Users\\xarthur\\AppData\\Local\\Temp\\pip-install-c20v04d_\\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\xarthur\AppData\Local\Temp\pip-record-jwqs25f2\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\xarthur\source\tmprepo\pyelastica\include\site\python3.9\llvmlite'
         cwd: C:\Users\xarthur\AppData\Local\Temp\pip-install-c20v04d_\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\
    Complete output (28 lines):
    running install
    running build
    got version from file C:\Users\xarthur\AppData\Local\Temp\pip-install-c20v04d_\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\llvmlite/_version.py {'version': '0.34.0', 'full': 'c5889c9e98c6b19d5d85ebdd982d64a03931f8e2'}
    running build_ext
    c:\users\xarthur\source\tmprepo\pyelastica\scripts\python.exe C:\Users\xarthur\AppData\Local\Temp\pip-install-c20v04d_\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\ffi\build.py
    CMake Error at CMakeLists.txt:3 (project):
      Generator

        Visual Studio 15 2017 Win64

      could not find any instance of Visual Studio.

    -- Configuring incomplete, errors occurred!
    See also "C:/Users/xarthur/AppData/Local/Temp/tmpvduc6zx7/CMakeFiles/CMakeOutput.log".
    Trying generator 'Visual Studio 15 2017 Win64'
    Traceback (most recent call last):
      File "C:\Users\xarthur\AppData\Local\Temp\pip-install-c20v04d_\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\ffi\build.py", line 191, in <module>
        main()
      File "C:\Users\xarthur\AppData\Local\Temp\pip-install-c20v04d_\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\ffi\build.py", line 179, in main
        main_win32()
      File "C:\Users\xarthur\AppData\Local\Temp\pip-install-c20v04d_\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\ffi\build.py", line 88, in main_win32
        generator = find_win32_generator()
      File "C:\Users\xarthur\AppData\Local\Temp\pip-install-c20v04d_\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\ffi\build.py", line 84, in find_win32_generator
        raise RuntimeError("No compatible cmake generator installed on this machine")
    RuntimeError: No compatible cmake generator installed on this machine
    error: command 'c:\\users\\xarthur\\source\\tmprepo\\pyelastica\\scripts\\python.exe' failed with exit code 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'c:\users\xarthur\source\tmprepo\pyelastica\scripts\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\xarthur\\AppData\\Local\\Temp\\pip-install-c20v04d_\\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\\setup.py'"'"'; __file__='"'"'C:\\Users\\xarthur\\AppData\\Local\\Temp\\pip-install-c20v04d_\\llvmlite_f3ca8c63d35f47e895df5b42bc8946cb\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\xarthur\AppData\Local\Temp\pip-record-jwqs25f2\install-record.txt' --single-version-externally-managed --compile --install-headers 'c:\users\xarthur\source\tmprepo\pyelastica\include\site\python3.9\llvmlite' Check the logs for full command output.
```

**To Reproduce**
```python install pyelastica
```

**Environment**
- Python/pip: 3.9/21.3.1
- OS, Device: Windows

**Expected behavior**
package successfully installed.

**Additional context**
It seems the installation requires a `Visual Studio 15 2017` compiler.
Since `VS 2022` has been released already, it would be helpful to add at least `Visual Studio 2019` into the list...

skim0119, 2022-01-18:
Hello @xarthurx
Thank you for addressing the issue.

I was able to replicate the same error on `Windows 10` with `Python 3.9`, when I did `pip install pyelastica`.

The current version of PyElastica does not support `Python 3.9`, because we are not sure if `llvmlite` (dependency for `numba`) supports `Python 3.9` yet. I successfully installed the package with `Python 3.8` on my machine. Could you try to downgrade your python version and retry installation?

xarthurx, 2022-01-18:
Confirmed.
python 3.8 works. 

Thank you for the info.