GazzolaLab/PyElastica #45: Inconsistent Numpy dependency version for Numba

bhosale2, 2022-01-22:
**Problem description**
Currently, the `requirements.txt` in PyElastica states the required version of `numpy` as `>=1.19.2` and `numba` as `==0.51.0`. However this seems to be inconsistent with `numba` [release notes](https://github.com/numba/numba/releases), which state that even the latest version of `numba` i.e. 0.55.0  doesnt support `numpy` latest release 1.22.1 and only has support till `numpy` version 1.21. While `numba` installs its own compatible `numpy` during installation, an update to `numpy` by the user can lead to dependency issues with `numba`.

**Suggested solution**
An upper limit on installation version of `numpy` i.e. `>=1.19.2,<1.22` in `requirements.txt` and other related locations (`setup.py`). An additional suggestion is update the `numba` dependency to 0.55.0, since all tests pass in PyElastica, with the latest `numba` version installed (on my Mac).

@armantekinalp  please confirm the above, after which I can take this issue.

skim0119, 2022-01-22:
If I remember correctly, there was a reason for fixing `numba version` to be 0.51.0. They moved `jitclass` into the experimental module, and we had some bug in the code. If we want to bump-up the version, we should check the codebase where we use `jitclass`.

I'm going to put this in `0.2.1` milestone. I think it should be easy fix.

bhosale2, 2022-01-22:
@skim0119 I just ran a search for `jitclass` across all files in PyElastica using `grep -R 'jitclass' .` and it yielded no matches, for both the branches present. Let me know if it checks out.

skim0119, 2022-01-22:
@bhosale2 Looks like it was in `_elastica_numba/_timestepper/_explicit_steppers` and `_elasstica_numba/_timestepper/_symplectic_steppers` in version `0.1.0.post1`. I don't see any trace after `0.1.0.post2`.

- We can change numba support to be `numba>=0.51.0`.
- I'm questioning if we need to put upper bound on `numpy` version. Pip should automatically handle numpy version that is compatible to numba, right? This should be handled by `numba`, and I don't think our code is incompatible to recent version of numpy.

bhosale2, 2022-01-23:
- Yes, when installed simultaneously using `requirements.txt`, `numba` takes care of the `numpy` dependency version as mentioned in the initial issue description. 
- However, I suggest adding the upper bound in the case when say the user installs another package that upgrades `numpy` in the same environment (happened in my case), which would then eventually lead to errors due to dependency mismatch in PyElastica. This would then have to be resolved by either downgrading `numpy` or reinstalling `numba`.
- To inform the user of the actual dependency limitations and to avoid any such hassles, I felt we should keep the dependency version consistent with that of `numba` release notes (they do the same on the [PyPI page](https://pypi.org/project/numba/)). 
- Yes, I feel also that our code is compatible with recent version of `numpy`.

skim0119, 2022-01-23:
I understand the issue, but I'm still not sure if we need to inform the dependency limitation that is not ours.

@bhosale2 Can you specify which package you used? If I understand pip correctly, installing numba (after installing an upper version of NumPy) should automatically downgrade NumPy to the compatible version. The error should happen when you force NumPy to upgrade.

My point is the error message should show that it is numba that is incompatible, not pyelastica. Either way, the user needs to downgrade NumPy to resolve the issue (I believe pip automatically does it).

Besides, the most recent version of NumPy is 1.22.1. I feel like numba will catch up within couple of weeks.

bhosale2, 2022-01-23:
I see I agree with all of the above and if `numba` will catch up in a few weeks, then we don't need to worry about putting an upper bound on `numpy`. I guess so we can close this issue after `numba` is bumped up as you mentioned.

skim0119, 2022-01-25:
The `numba` version is updated in 5c31cfc0ad9e50fd2b0e05b926aa0d34644ff929 and 9bcfd0f050f3ce8fbd5995b65827397693e27544.

closed 2022-01-25 by skim0119