GazzolaLab/PyElastica #52: Recommonmark is deprecated: We should switch to MyST

skim0119, 2022-02-07:
## Context
We are currently using `sphinx` and `ReadtheDocs` to keep our documentation clean and organized. Natively, `sphinx` uses `reStructuredText` (or .rst extension), and the module `recommonmark` has been used to parse `markdown` format. I ​recently found that the module `recommonmark` is no longer maintained, and the developer (and community) shifted to `MyST` parser.

##  Suggestion
I don't think the issue is imminent and we can probably handle the problem in the back, but I can definitely see the community and the supports for the `markdown` + `sphinx` are shifting to `MyST`. It appears to me that `MyST` is now officially supported `markdown` extension for `sphinx`, and we should change our environment as well.

### Change
It looks like `MyST` is fairly easy to use just like `recommonmark`. Most of the change will be done on `conf.py`, and everything should be compatible.

## Notes

[Deprecation note from recommonmark](https://github.com/readthedocs/recommonmark/issues/221)

skim0119, 2022-02-07:
Working on my [local branch](https://github.com/skim0119/PyElastica/tree/docs_myst).

I'm going to run few test and make sure documentations are properly generated, and merge to `doc_patch` branch.

nmnaughton, 2022-02-07:
Just so I am clear, with this patch we would need to convert all the *.rst files to *.md? (It looks like we can do that automatically.) I think having all of the docs page in markdown would work better anyway, so this seems like a  worthwile patch to push through.

skim0119, 2022-02-07:
@nmnaughton We can still use rst. We don't need to convert.

PR #53 (2022-02-07): Documentation updates using MyST
## Changes

- Remove `recommonmark` and add `MyST` (resolve #52)
  - Add `numpydoc` for parsing numpy-documentation style.
- Rework docs/readme.md file for the developer, including explanation on how `documentations` are made.
- organize documentation files (API, overview)
- minor changes in the class/method description to match `MyST` and `numpydoc`.
closed 2022-02-07 by skim0119