GazzolaLab/PyElastica #68: #GSoC Project 5: Debugger/logger of simulation instabilities

brlrb, 2022-03-09:
Hi PyElastica Team,

I am interested in GSoC 2022 and specifically `Project 5: Debugger/logger of simulation instabilities`. 

I want to clarify a few things:
- is the task to create a logger which binds to the simulation? 
- should the crash/unstable log be a textfile or based on HTML UI 

Thank you! I am looking forward to what I can contribute.

nmnaughton, 2022-03-09:
Hi @brlrb, thanks for your interest in the project!

1. We have some logging/callback capabilities already implemented that it would be nice to incorporate to avoid duplicated code. I think we are envisioning a logging subroutine that can be enabled by the user to track down instabilities as opposed to something the binds or wraps the simulation, however, we are flexible on this.
2. I think it would need to be both. Simulations generate a lot of data that need to be parsed after the fact to understand what happened, so a textfile/csv/database file will be needed to store the data, however a HTML UI to view that data would be a very nice feature! Something like tensorboard comes to mind...

nmnaughton, 2022-04-28:
GSoC submissions are now closed. We did not receive a proposal for this project.

closed 2022-04-28 by nmnaughton