GazzolaLab/PyElastica #81: Dissipation constant should be scaled with mass instead of lengths

armantekinalp, 2022-05-11:
**Describe the bug**
In Elastica `dissipation constant (nu)` is defined as per unit length  and its unit is `kg/(m.s)`. However, this assumes that rods are uniform. However, in the case of tapered rods `dissipation constant` should also be scaled with the area otherwise we will be applying same dissipation force for two elements that have same velocity but different mass. 

**How to fix**
1. Define the dissipation constant as `1/s` .
2. In `allocate` function,  scale the user defined `nu` with `element mass` and store it.
3. Remove multiplication by `lengths` while computing the dissipation forces and torques.
https://github.com/GazzolaLab/PyElastica/blob/fee87b9da22e310ff925c16fdc839bf8405c51a4/elastica/rod/cosserat_rod.py#L748
https://github.com/GazzolaLab/PyElastica/blob/fee87b9da22e310ff925c16fdc839bf8405c51a4/elastica/rod/cosserat_rod.py#L843

bhosale2, 2022-05-12:
Hi @armantekinalp some suggestions based on changes in `elasticapp`. 

1. The name of the parameter must first be changed to `damping_rate` owing to its nature and units `1/s`.
2.  Instead of multiplying in `allocate` by mass, we should set it as it is there, and for `velocity` and `omega` multiply by `mass` and `J / e` for `omega` when computing damping forces.
3. A note should be made in the example cases for damping, where we need to rescale these values to match the old implementation effect.

bhosale2, 2022-06-01:
Closed by #87.

PR #87 (2022-06-01): [WIP] (#81) dissipation constant fix
Fixes #81, as per our discussion @armantekinalp this PR is WIP and we will merge once the dissipation constants are rescaled for examples, and all examples with `nu` have been confirmed to run fine. Other than that, please go through PR and let me know if any changes are needed.
closed 2022-06-01 by bhosale2