GazzolaLab/PyElastica #99: Unit tests for contact module

armantekinalp, 2022-06-01:
**Is your feature request related to a problem? Please describe.**
Although we have tested contact module through different simulations, there is no unit tests for contact module. This can cause vulnerability in development process as well as reduces code coverage. 

**Describe the solution you'd like**
Implement unit tests for contact module.

armantekinalp, 2023-05-09:
Here are the list of functions that needs to be tests

- Common functions (1 PR)
    - [ ] ` _dot_product`
    - [ ] `_norm`
    - [ ] `_clip`
    - [ ] `_out_of_bounds`
    - [ ] `_find_min_dist`
    - [ ] `_aabbs_not_intersecting`

- More specific functions (2-3 PR)
    - [ ] `_prune_using_aabbs_rod_rigid_body`
    - [ ] `_prune_using_aabbs_rod_rod`
    - [ ] `_calculate_contact_forces_rod_rigid_body`
    - [ ] `_calculate_contact_forces_rod_rod`
    - [ ] `_calculate_contact_forces_self_rod`
    - [ ] 
- Wrapper classes (2 PR)
    - [ ] `ExternalContact`
    - [ ] `SelfContact`

Rahul-JOON, 2023-06-03:
The specific functions are now complete, I will do the wrapper classes next.

armantekinalp, 2023-06-12:
This issue is resolved, I am closing.

closed 2023-06-12 by armantekinalp