GazzolaLab/PyElastica #105: Simulating dynamics of a pre curved rod straightening out

itiwari93, 2022-06-05:
Hello PyElastica team,
I have a problem that I am trying to simulate using PyElastica. I am trying to simulate the dynamics of a pre-curved rod opening-up. But I am currently struggling with the stability of the solution, as it keeps blowing up to NaN values.  I pre-curved the rod using the instructions in issue#12 of this project. I intentionally did not change the resting curvature of the rod to the pre-curved configuration, so that the rod tends to straighten itself as soon as the simulation starts. I expect the rod to at least straighten up but the simulation breaks down into NaN values.

I just wanted to ask for help here, in case anyone notices some basic novice-level mistake I might be making. I am very new to using PyElastica and even the field of Cosserat rod simulation, so please forgive me if this code is not up to the mark. I am sharing it below. I will be very thankful if anyone could point me in the right direction. 
By the way, this is a great project, cannot wait for more enhancements and updates!
```

import numpy as np

# import wrappers
from elastica.wrappers import BaseSystemCollection, Constraints, Forcing, CallBacks

# import rod class and forces to be applied
from elastica.rod.cosserat_rod import CosseratRod

# import timestepping functions
from elastica.timestepper.symplectic_steppers import PositionVerlet
from elastica.timestepper import integrate

# import call back functions
from elastica.callback_functions import CallBackBaseClass
from collections import defaultdict

#%%
n_nodes=30
n_elem=n_nodes-1
positions=np.zeros([3,n_nodes])
directors=np.zeros([3,3,n_elem])
x=np.linspace(0,10,n_nodes)
y=x**2

# from matplotlib import pyplot as plt

# plt.plot(x,y)

ii=0
positions[0,:]=x
positions[1,:]=y
positions[2,:]=0

d3=np.zeros([1,3,n_elem])
tempx=np.diff(x)
tempy=np.diff(y)
tempmod=np.sqrt(tempx**2+tempy**2)
rod_len=np.sum(tempmod)
tempx=tempx/tempmod
tempy=tempy/tempmod
directors[2,0,:]=tempx
directors[2,1,:]=tempy
directors[1,2,:]=-np.ones(np.shape(tempx))
directors[0,0,:]=directors[1,1,:]*directors[2,2,:]-directors[1,2,:]*directors[2,1,:]
directors[0,1,:]=-(directors[1,0,:]*directors[2,2,:]-directors[1,2,:]*directors[2,0,:])

#%%

class RodSimulator(BaseSystemCollection, Constraints, Forcing, CallBacks):
    pass

Rod_sim = RodSimulator()

# Define rod parameters
n_elem = n_nodes-1
start = np.array(positions[:,0])
direction = np.array([0.1, 0.0, np.sqrt(1-0.1**2)])
normal = np.array([0.0, 1.0, 0.0])
base_length = rod_len  #cm    
base_radius = 0.1   #cm
base_area = np.pi * base_radius ** 2
density = 19.250   #this is in g/(cc)
nu = 180                 #gram/cm.second
E = 3.4*10**12   #this is not in Pascal but gram/(cm.second^2)
poisson_ratio = 0.5
shear_modulus = E / (poisson_ratio + 1.0)

# Create rod
shearable_rod = CosseratRod.straight_rod(
    n_elem,
    start,
    direction,
    normal,
    base_length,
    base_radius,
    density,
    nu,
    E,
    shear_modulus=shear_modulus,
    position=positions, 
    directors=directors
)

aa=shearable_rod.director_collection
# Add rod to the snake system
Rod_sim.append(shearable_rod)

#%%

period=1   #second   
print(shearable_rod.compute_bending_energy())

# Add call backs
class ContinuumRodCallBack(CallBackBaseClass):
    """
    Call back function for continuum snake
    """

    def __init__(self, step_skip: int, callback_params: dict):
        CallBackBaseClass.__init__(self)
        self.every = step_skip
        self.callback_params = callback_params

    def make_callback(self, system, time, current_step: int):

        if current_step % self.every == 0:

            self.callback_params["time"].append(time)
            self.callback_params["step"].append(current_step)
            self.callback_params["position"].append(system.position_collection.copy())
            self.callback_params["velocity"].append(system.velocity_collection.copy())
            self.callback_params["avg_velocity"].append(
                system.compute_velocity_center_of_mass()
            )

            self.callback_params["center_of_mass"].append(
                system.compute_position_center_of_mass()
            )
            self.callback_params["curvature"].append(system.kappa.copy())

            return

pp_list = defaultdict(list)
Rod_sim.collect_diagnostics(shearable_rod).using(
    ContinuumRodCallBack, step_skip=1000, callback_params=pp_list
)
print("Callback function added to the simulator")

#%%

Rod_sim.finalize()

final_time = 1  #second
dt = 1.0e-6    #second
total_steps = int(final_time / dt)
print("Total steps", total_steps)

timestepper = PositionVerlet()

integrate(timestepper, Rod_sim, final_time, total_steps)

#%%

from IPython.display import Video

def plot_video(plot_params: dict, video_name="video.mp4", margin=0.2, fps=15):
    from matplotlib import pyplot as plt
    import matplotlib.animation as manimation
    from mpl_toolkits import mplot3d

    t = np.array(plot_params["time"])
    positions_over_time = np.array(plot_params["position"])
    total_time = 5#int(np.around(t[..., -1], 1))
    total_frames = fps * total_time
    step = round(len(t) / total_frames)
    print("creating video -- this can take a few minutes")
    FFMpegWriter = manimation.writers["ffmpeg"]
    metadata = dict(title="Movie Test", artist="Matplotlib", comment="Movie support!")
    writer = FFMpegWriter(fps=fps, metadata=metadata)
    fig = plt.figure()
    ax = fig.add_subplot(111, projection="3d")
    #ax.set_xlim(0 - margin, 3 + margin)
    #ax.set_ylim(-1.5 - margin, 1.5 + margin)
    #ax.set_zlim(0, 1)
    ax.view_init(elev=90, azim=-90)
    ax.set_ylabel("Y Position (cm)", fontsize=12)
    ax.set_xlabel("X Position (cm)", fontsize=12)
    ax.set_zlabel("Z Position (cm)", fontsize=12)
    rod_lines_3d = ax.plot(
        positions_over_time[0][0],
        positions_over_time[0][1],
        positions_over_time[0][2],
        linewidth=3,
    )[0]
    with writer.saving(fig, video_name, dpi=100):
        with plt.style.context("seaborn-whitegrid"):
            for time in range(1, len(t), step):
                rod_lines_3d.set_xdata(positions_over_time[time][0])
                rod_lines_3d.set_ydata(positions_over_time[time][1])
                rod_lines_3d.set_3d_properties(positions_over_time[time][2])

                writer.grab_frame()
    plt.close(fig)

filename_video = "continuum_Rod_3d.mp4"
plot_video(pp_list, video_name=filename_video, margin=0.2, fps=60)

Video("continuum_rod_3d.mp4")

#%%

def plot_Rod(plot_params: dict,shearable_rod):
    import matplotlib.pyplot as plt
    t = np.array(plot_params["time"])
    positions_over_time = np.array(plot_params["center_of_mass"])
    fig = plt.figure(figsize=(5, 4), frameon=True, dpi=150)
    ax = fig.add_subplot(111)
    ax.grid(b=True, which="major", color="grey", linestyle="-", linewidth=0.25)
    ax.plot(
        t[:],
        positions_over_time[:,2],
        "b*",
        label="n=" + str(shearable_rod.n_elems),
    )

    ax.legend(prop={"size": 12})
    ax.set_ylabel("Z Position (cm)", fontsize=12)
    ax.set_xlabel("Time (second)", fontsize=12)
    plt.show()

plot_Rod(pp_list,shearable_rod)

```

armantekinalp, 2022-06-07:
Hi @itiwari93 

I ran your code, and I think your problem is too stiff and `nu` might be too large. Do you use values from literature?  In any case I am attaching some values and you can use and start tuning them for your problem.

```python
# Define rod parameters
n_elem = n_nodes - 1
start = np.array(positions[:, 0])
direction = np.array([0.1, 0.0, np.sqrt(1 - 0.1 ** 2)])
normal = np.array([0.0, 1.0, 0.0])
base_length = rod_len  #cm
base_radius = 0.1  #cm
base_area = np.pi * base_radius ** 2
density = 19.250  #this is in g/(cc)
nu = 180/10000  #gram/cm.second
E = 1E5#3.4 * 10 ** 12  #this is not in Pascal but gram/(cm.second^2)
poisson_ratio = 0.5
shear_modulus = E / (poisson_ratio + 1.0)

```

If you need to deal with rod's that have `E=1E12`, then one other suggestion is the following. You can set pre-curved condition as the `rest_kappa` and `rest_sigma` for your rod. Then you can write an `ExternalForce` class and slowly ramp up `rest_kappa` and `rest_sigma` to your desired condition. If you are interested in capturing dynamics immediately after releasing then you should be careful and choose a smaller ramp up time.

Let me know if this helps.

itiwari93, 2022-06-07:
Thank you for your answer @armantekinalp . I took these values from some literature about metal rods. I will definitely try creating the `ExternalForce` class to make the rod straighten up at a small enough timescale so that I am able to study some qualitative dynamics without breaking the simulation. I also tried rescaling the time and increasing `nu` to large values, but that does not help beyond a point.
I will get back when I have tried the possible solutions you suggested.
Thank you once again!

itiwari93, 2022-06-18:
Hi @armantekinalp,
Thank you for your suggestion last time. 
I tried the following protocol, stored the `bend_matrix` and `shear_matrix` of the Cosserat rod in a separate variable, then put
the actual `shearable_rod.bend_matrix = 0` and `shearable_rod.shear_matrix = 0`. The idea was to slowly increase both these
matrices to their true value while the simulation ran. Unfortunately, I was unable to change these two properties of the rod while the simulation ran, so I came up with the following external force function by patching together codes/functions from the internal force and torque calculation that happens due to the curvature of the rod.  I attach the code snippet below:

`

class straighten_slowly(NoForces):

    def __init__(self, ramp_up_time, zero_kappa,zero_sigma,
                 og_bend_matrix, og_shear_matrix):
        #self.k = k
        self.ramp_up_time = ramp_up_time
        self.zero_kappa = zero_kappa
        self.zero_sigma = zero_sigma
        self.og_bend_matrix = og_bend_matrix
        self.og_shear_matrix = og_shear_matrix 
    
    def apply_forces(
        self,
        system,
        time: np.float = 0.0):
        factor = min(time/self.ramp_up_time, 1.0)
        position_collection = system.position_collection
        volume = system.volume
        lengths = system.lengths
        tangents = system.tangents
        radius = system.radius
        rest_lengths = system.rest_lengths
        rest_voronoi_lengths = system.rest_voronoi_lengths
        dilatation = system.dilatation
        voronoi_dilatation = system.voronoi_dilatation
        director_collection = system.director_collection
        sigma = system.sigma
        rest_sigma = self.zero_sigma
        shear_matrix = self.og_shear_matrix
        director_collection = system.director_collection
        rest_voronoi_lengths = system.rest_voronoi_lengths
        bend_matrix = self.og_bend_matrix
        kappa = system.kappa
        rest_kappa = self.zero_kappa
        internal_couple=system.internal_couple
        internal_stress=system.internal_stress
        """
        Update <internal stress> given <shear matrix, sigma, and rest_sigma>.
        Linear force functional
        Operates on
        S : (3,3,n) tensor and sigma (3,n)
        """
        _compute_shear_stretch_strains(
            position_collection,
            volume,
            lengths,
            tangents,
            radius,
            rest_lengths,
            rest_voronoi_lengths,
            dilatation,
            voronoi_dilatation,
            director_collection,
            sigma,
        )
        # system.external_forces += factor * _batch_matvec(shear_matrix, sigma - rest_sigma)
        #inplace_addition(system.external_forces, factor * _batch_matvec(shear_matrix, sigma - rest_sigma))
        internal_stress[:] = factor * _batch_matvec(shear_matrix, sigma - rest_sigma)
        
        
    # def _compute_internal_bending_twist_stresses_from_model(
    #     self, system, 
[... 69 lines cut ...]
        bend_twist_couple_3D = quadrature_kernel_for_block_structure(
            _batch_cross(kappa, internal_couple)
            * rest_voronoi_lengths
            * voronoi_dilatation_inv_cube_cached,
            system.ghost_voronoi_idx,
        )
        # (Qt x n_L) * \hat{l}
        shear_stretch_couple = (
            _batch_cross(_batch_matvec(director_collection, tangents), internal_stress)
            * rest_lengths
            )
        blocksize = system.internal_torques.shape[1]
        for i in range(3):
            for k in range(blocksize):
                system.external_torques[i, k] += (
                    bend_twist_couple_2D[i, k]
                    + bend_twist_couple_3D[i, k]
                    + shear_stretch_couple[i, k]
                )
`

Unfortunately again, I am stuck at another problem here, I cannot seem to access the rod variable ghost_elems_idx. 

Is there anything obviously incorrect in my approach which I can change? Or is there a better way to accomplish the same thing? Any pointers in the right direction will be very much appreciated. 
I thank you for your time and patience.

armantekinalp, 2022-06-21:
Hi @itiwari93 ,

Why did you set `shearable_rod.bend_matrix=0` and `shearable_rod.shear_matrix=0`  after initializing the rod? There are two issues for this assignment.

1.  `bend_matrix` and `shear_matrix` are arrays when you do the above operation you convert them to floats which breaks a lot of references at the backhand. If you really want to set them to `zero` do in-place operation `shearable_rod.bend_matrix[:]=0` and `shearable_rod.shear_matrix[:]=0` 
2. I am also confused why you set these arrays to zero in terms of physics. They are material properties so they should be set in the begging of the simulation and kept constant. Unless you have a specific problem that material properties changes due to some environment factor i.e. heating the rod etc.

Previously what I suggested is to start with pre-curved rod and change the `rest_kappa` and `rest_sigma` slowly to desired curvature and strain. Something like below

```python

class UpdateRestKappaSigma(NoForces):
def __init__(self, ramp_up_time, initial_kappa, initial_sigma, target_kappa, target_sigma):
      self. ramp_up_time = ramp_up_time
      self. initial_kappa = initial_kappa
      self. initial_sigma = initial_sigma
      self. target_kappa = target_kappa
      self.target_sigma = target_sigma

def apply_forces(self, system, time: np.float = 0.0):
      factor = min(time/self.ramp_up_time, 1.0)
      system.rest_kappa[:] = self.initial_kappa - factor * (self.initial_kappa - self.target_kappa)
      system.rest_sigma[:] = self.initial_sigma - factor * (self.initial_sigma - self.target_sigma)
``` 

Let me know this make sense

itiwari93, 2022-06-21:
Hi @armantekinalp,

I apologise for the confusing phrasing of my last message. When I said that I had put `shear_matrix` and `bend_matrix` to zero, I meant I had put them elementwise to zero.  To be honest, I initially struggled to change the `rest_kappa` using the `External_force` method, therefore I shifted to the approach of changing `bend_matrix` instead of `rest_kappa`.
The physical reasoning to do that is as follows:
I imagined the shear response and bending response of the rod as some abstract 
spring where for example, `F_bend = -k_bend (x-x_0)` ,  where `k_bend` corresponded to the `bend_matrix` and `x_0` corresponded to the `rest_kappa` in some abstract space.
Therefore, instead of slowly moving  `x_0` to the desired value, I decided to slowly increase `k_bend` from 0 to its correct value. I was hoping the two approaches should be identical? 
I was actually able to access the `ghost_elems_idx` values and make the simulation work by calling the function `make_block_memory_metadata` function in the `memory_block_rod` script of PyElastica.

All said, I really appreciate all the time and effort you have put to answer my naive questions and to the whole Elastica team for making this awesome project. I look forward to using this even more and reading about more interesting works coming out of this project.
Thank you. :-D

bhosale2, 2022-07-01:
Closing this since resolved.

closed 2022-07-01 by bhosale2