GazzolaLab/PyElastica #106: Ramp up time default to 0.0 for EndpointForces

Jarvis-X, 2022-06-06:
**Describe the bug**
It should be a quick fix: the default value of `ramp_up_time=0.0` causes divided-by-zero error when initializing.

**To Reproduce**
```py
# any rod created for any simulation
mySim.add_forcing_to(rod).using(
EndpointForces, np.array([0, 0, 0]), np.array([-10, 0, 0]))

mySim.finalize()
timestepper = PositionVerlet()

# some time-step related parameters
integrate(timestepper, mySim, final_time, total_steps)
```
```bash
---------------------------------------------------------------------------
ZeroDivisionError                         Traceback (most recent call last)
[<ipython-input-8-285f38790693>](https://localhost:8080/#) in <module>()
    135 # timestepper = PEFRL()
    136 
--> 137 integrate(timestepper, pendulum_sim, final_time, total_steps)
    138 
    139 if PLOT_FIGURE:

4 frames
[/usr/local/lib/python3.7/dist-packages/elastica/timestepper/__init__.py](https://localhost:8080/#) in integrate(StatefulStepper, System, final_time, n_steps, restart_time, progress_bar, **kwargs)
    111 
    112     for i in tqdm(range(n_steps), disable=(not progress_bar)):
--> 113         time = do_step(StatefulStepper, stages_and_updates, System, time, dt)
    114 
    115     print("Final time of simulation is : ", time)

[/usr/local/lib/python3.7/dist-packages/elastica/timestepper/symplectic_steppers.py](https://localhost:8080/#) in do_step(TimeStepper, _steps_and_prefactors, SystemCollection, time, dt)
     89 
     90             # Add external forces, controls etc.
---> 91             SystemCollection.synchronize(time)
     92 
     93             for system in SystemCollection._memory_blocks:

[/usr/local/lib/python3.7/dist-packages/elastica/wrappers/base_system.py](https://localhost:8080/#) in synchronize(self, time)
    157         # Collection call _featuer_group_synchronize
    158         for feature in self._feature_group_synchronize:
--> 159             feature(time)
    160 
    161     def constrain_values(self, time: float):

[/usr/local/lib/python3.7/dist-packages/elastica/wrappers/forcing.py](https://localhost:8080/#) in _call_ext_forces_torques(self, time, *args, **kwargs)
     83     def _call_ext_forces_torques(self, time, *args, **kwargs):
     84         for sys_id, ext_force_torque in self._ext_forces_torques:
---> 85             ext_force_torque.apply_forces(self._systems[sys_id], time, *args, **kwargs)
     86             ext_force_torque.apply_torques(self._systems[sys_id], time, *args, **kwargs)
     87             # TODO Apply torque, see if necessary

[/usr/local/lib/python3.7/dist-packages/elastica/external_forces.py](https://localhost:8080/#) in apply_forces(self, system, time)
    172             self.end_force,
    173             time,
--> 174             self.ramp_up_time,
    175         )
    176 

ZeroDivisionError: division by zero
```

**Environment**
- Python/pip version 21.1.3
- Numpy/numba version 0.51.2
- PyElastica version 0.2.3
- OS, Device Google Colab

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

skim0119, 2022-06-07:
@Jarvis-X Thank you for pointing it out. We'll include the change in the next version `v0.2.4`.

skim0119, 2022-06-08:
The update is included in the `update-0.2.4` branch (a511e49)

commit a511e49a (2022-06-08): resolve #106: avoid 0.0 default for rampup