GazzolaLab/PyElastica #107: Adding Interaction plane to slithering snake example

matei1996, 2022-06-10:
Dear PyElastica Team,

I am currently trying to add an Interaction plane (or any arbitrary obstacle) to the slithering snake example, so that the snake "bounces of" the plane and continues to move into a different direction. However, I wasn't able to find out the correct way to do it. I tried adding the following code into the example

side_plane = np.array([0.0, 0.0, 3.0])
side_normal = np.array([0.0, 0.0, 1.0])
surface_tol=1e-4

plane_sim = InteractionPlane(k=1.0,nu=1e-6,plane_origin=side_plane,plane_normal=normal_plane)

I do not obtain any error messages. Unfortunately, the code does not seem to have any effect on the snake. I am thankful for any tips. I am very new to the PyElastica (and Github) implementation, however I think this is a great project!

armantekinalp, 2022-06-12:
Hi @matei1996 

You cannot add  `InteractionPlane` class to the `simulator` since it is parent class of `AnisotropicFrictionalPlane`. Because of that you don't see any difference in your simulation with/without `InteractionPlane`.  

If you want to add obstacles, you can add rigid cylinders and set contact between rod and cylinders. [Here](https://github.com/GazzolaLab/PyElastica/tree/update-0.2.4/examples/RigidBodyCases/RodRigidBodyContact) you can see the examples how to do.

Let me know if this helps.

matei1996, 2022-06-14:
Hi @armantekinalp 

Thanks a lot. Yes this helps me for now. Is it also possible to use something else than a cylinder as obstacle? I would like to into interaction with a plane or letting the rod start inside the cylinder.

Best regards,
Matei

armantekinalp, 2022-06-14:
No, currently we only support contact between rod-cylinder and rod with flat plane (AnisotropicFrictionalPlane). However, you can extend the contact and interaction classes based on your problem requirements.

matei1996, 2022-06-16:
Ok, thank you! One other question. Is it possible to integrate multiple cylinders? Through my first attempts it seems to be the case. However, I still have problems with the plot_cylinder_rod_position, since it only takes as argument cylinder_history of 1 cylinder. I tried using further "plot_cylinder_rod_position" functions with different cylinders but that didn't work as intended. Can I use only one function for all cylinders?

matei1996, 2022-06-21:
Hi again,

I am also trying to rotate the cylinder by the changing the normal and the directional vector, however it does not seem work. Do you have any idea why this might be the case? If I should open a new issue please let me know. I am unsure...

Best regards,
Matei

armantekinalp, 2022-06-21:
> Ok, thank you! One other question. Is it possible to integrate multiple cylinders? Through my first attempts it seems to be the case. However, I still have problems with the [...]

You need to define `callback` functions for all cylinders that are `append` to the simulation, so you can record their time history. For plotting you need to update the plotting scripts.

matei1996, 2022-06-22:
Thanks. I was able to plot multiple rods now.

Regarding the position of the cylinder: Right not I am only trying to change the direction and normal of the rod at initialization. However, this does not work yet as intended. The cylinders do not change compared to the original position in the example. 
(https://github.com/GazzolaLab/PyElastica/issues/107#:~:text=rod%20and%20cylinders.-,Here,-you%20can%20see)

Furthermore, I noticed that some specific interactions of the rod lead to the following error message:

<ipython-input-236-e7bd12932f48>:95: UserWarning: Z contains NaN values. This may result in rendering artifacts.
  surf = ax.plot_surface(XC, YC, ZC, color="g", alpha=0.5)

I assume it depends on the contact position of the rod with the cylinder. Usually the error occurs if the rod is hitting the cylinder "further to the middle" and therefore is deflected stronger as opposed to hitting it on the upper or lower edges.
Is there an explanation for this?

armantekinalp, 2022-06-22:
> Regarding the position of the cylinder: Right not I am only trying to change the direction and normal of the rod at initialization. However, this does not work yet as intended. The cylinders do not change compared to the original position in the example. (#107)

Can you share me how you initialize the cylinder? Cylinder should rotate but the plotting might have issues. 

> I assume it depends on the contact position of the rod with the cylinder. Usually the error occurs if the rod is hitting the cylinder "further to the middle" and therefore is deflected stronger as opposed to hitting it on the upper or lower edges. Is there an explanation for this?

Do you have any videos or some plots? I am currently working on improving the rod-cylinder contact model and there should be a PR soon. Maybe that can fix the problem you have.

matei1996, 2022-06-23:
To initialize the cylinder i use the same code that is used in the example you mentioned above. (https://github.com/GazzolaLab/PyElastica/issues/107#:~:text=rod%20and%20cylinders.-,Here,-you%20can%20see)

I initialize it using

cylinder_start = np.array([0.3, 0.0, 0.0])
cylinder_direction = np.array([0.0, 1.0, 0.0])
cylinder_normal = np.array([0.0, 0.0, 1.0])
cylinder_height = 0.9
cylinder_radius = 10.0 * base_radius
density_rod=10

cylinder = Cylinder(
    cylinder_start,
    cylinder_direction,
    cylinder_normal,
    cylinder_height,
    cylinder_radius,
    density_rod,
)

and then change the parameters. I would like to have the cylinder pointing upwards.

For the second problem I can only attach a video where everything is working, when I get an error message, no video is created.

https://user-images.githubusercontent.com/87129296/175241557-0c73e5b0-35fc-4a84-a82e-dec80497aa36.mp4

matei1996, 2022-06-27:
Hi, I have one more different question. I don't know if I should open a new request. If yes please let me know.

We are trying right now to model a long rod that enters an environment(like the above one with multiple cylinders) through some narrow entry. So we would like to have some boundary conditions at the entrance, for example at the point (0,0), where the rod is fixed and can not change. Is there an example where something similar is done?

We would need moving BC where the values are time dependent, meaning we would like to have BC at a specific element of the rod and after some time shift the BC to the next element.
If something like this isn't possible yet, do you know which part of the source code has to be modified to achieve this result?

armantekinalp, 2022-06-28:
> I initialize it using
>
> cylinder_start = np.array([0.3, 0.0, 0.0]) cylinder_direction = np.array([0.0, 1.0, 0.0]) cylinder_normal = np.array([0.0, 0.0, 1.0]) cylinder_height = 0.9 cylinder_radius = [...]

It seems there is an issue in plotting script and not the initialization of cylinder.  You can check the tangent of cylinder as printing out the `cylinder.director_collection[2,:,0]`  and it should be different if you change ` cylinder_direction`.

armantekinalp, 2022-06-28:
> For the second problem I can only attach a video where everything is working, when I get an error message, no video is created.

It seems you have a numerical problem.  Have you tried the contact with smaller contact constant?

armantekinalp, 2022-06-28:
> Hi, I have one more different question. I don't know if I should open a new request. If yes please let me know.
>
> We are trying right now to model a long rod that enters an environment(like the above [...]

We don't have a boundary condition exactly for your needs so you need to write your own custom boundary condition class. I think [HelicalBucklingBC](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/boundary_conditions.py#L435) is a good example for you to start. This BC class slowly rotates the ends of the rods and pulls it  by changing the angular and translational velocity. So there are time-dependency, and modifying the position of end node/elements.

closed 2022-07-27 by matei1996