GazzolaLab/PyElastica #113: Rename joint.py to have a more intuitive name

bhosale2, 2022-06-16:
Currently, the implementation for connections and collision lies in `joint.py` which is a rather non-intuitive name for its contents. A more intuitive name could be `connections_and_collision.py`? @armantekinalp and @skim0119 suggestions?

skim0119, 2022-06-17:
I think `collision` should be separated. I agree that `connection could be more general name.

bhosale2, 2022-06-18:
Ok, then the consensus on this is:
1. Rename `joint.py` to `connections.py`
2. Move out collision classes into a separate `collision.py`

armantekinalp, 2022-06-18:
Yes, I think it is a good idea

Ali-7800, 2023-07-17:
Merging checklist
- [ ] New contact module and tests
- [ ] Updated rod-rod, rod-cylinder, self-contact classes, examples, and tests
- [ ] Surface class, surface subclass: plane, and tests
- [ ] Updated rod-plane, cylinder-plane classes, examples, and tests

Rahul-JOON, 2023-08-08:
@Ali-7800 added the New Contact module, surface base class, and tests via #281.

Ali-7800, 2023-08-21:
Added Updated rod-rod, rod-cylinder, self-contact classes, examples, and tests via #291

Ali-7800, 2023-10-27:
Added the new plane surface subclass.

Ali-7800, 2023-10-31:
Added updated rod-plane contact class

Ali-7800, 2023-11-07:
Updated rod-plane contact classes tests and refactored contact functions in interaction.py and joint.py

Ali-7800, 2023-12-06:
Updated rod-plane contact examples and added continuum snake with lifting wave example

Ali-7800, 2024-02-20:
Updated documentation for contact module, closing now as issue is resolved

closed 2024-02-20 by Ali-7800