GazzolaLab/PyElastica #129: Missing documentation on the use of Numba

mstoelzle, 2022-07-06:
**Describe the bug**
I was wondering whether the numba jit compilation is active by default or if I need to activate it manually somewhere.
I am not sure if I missed this information somewhere. If yes, please excuse me and close this issue.  Otherwise, it might be useful to add this information somewhere in the documentation.

armantekinalp, 2022-07-06:
Numba  is  a requirement of PyElastica and it is automatically activated. Thank you for the suggestion we can add this information in the documentation.

skim0119, 2022-07-06:
Most of the expansive operators and functions are already decorated with `numba`, and we dropped the `numpy-only` support since `v0.1.0`. Numba is applied and activated to all the core kernels.

We indicated the `numba` as a core dependencies in the documentation ([page](https://docs.cosseratrods.org/en/latest/overview/installation.html)).