GazzolaLab/PyElastica #132: Definition of normal direction for hinge joint in inertial instead of local frame

mstoelzle, 2022-07-12:
**Describe the bug**
As it can be see in the code below, the normal direction of the hinge joint is never transformed and directly used in the world frame. This means, that the normal direction needs to be specified in the inertial frame by the user. However, if both of the two systems connected by the hinge joint also move in 3D space, the rotating axis of the hinge stays constant in the world frame, which is not correct what happens in the real-world. We would expect the rotation axis of the hinge join to also move in 3D space together with the movement of both links / bodies / rods. Accordingly, the normal direction of the hinge joint should be specified in the local frame by the user and then at every time-step rotated into the world frame for the necessary computations.

**To Reproduce**
```py
 def apply_torques(self, system_one, index_one, system_two, index_two):
        # current direction of the first element of link two
        # also NOTE: - rod two is hinged at first element
        link_direction = (
                system_two.position_collection[..., index_two + 1]
                - system_two.position_collection[..., index_two]
        )

        # projection of the link direction onto the plane normal
        force_direction = (
                -np.dot(link_direction, self.normal_direction) * self.normal_direction
        )

        # compute the restoring torque
        torque = self.kt * np.cross(link_direction, force_direction)

        # The opposite torque will be applied on link one
        system_one.external_torques[..., index_one] -= (
                system_one.director_collection[..., index_one] @ torque
        )
        system_two.external_torques[..., index_two] += (
                system_two.director_collection[..., index_two] @ torque
        )
```

**Environment**
- Python=3.10.5, pip=22.1.2 
- Numpy/numba version: 22.1.2
- PyElastica version: 0.2.3
- OS, Device: macOS / macBook Pro 15'' 2019

**Expected behavior**
I would expect to be able to specify the normal direction of the hinge joint in the local coordinate frame of the first system, as the first system could be moving around in 3D space as well, and the functionality of the hinge would always stay constant with respect to the two local systems, instead of the world.

armantekinalp, 2022-07-13:
For the hinge joint intention was to demonstrate how to connect two rods (one fixed other free to move) at the their ends and restrict the motion in a plane. I think documentation is misleading here.

We will take care this issue in our next release, and implement an algorithm to constrain relative motions of rods even they move in 3D space. 

However, if you want/need this implementation soon I can explain the algorithm details here.

bhosale2, 2022-08-01:
@mstoelzle whats the status on this one?

mstoelzle, 2022-08-01:
> @mstoelzle whats the status on this one?

@bhosale2 Thanks for your message. The HingeJoint is not currently needed for my research. Thus, I am not currently working on an improved implementation. I just noticed the shortcomings of the current implementation while looking through the code.

At least the definition of the plane in the local instead the inertial frame should be very easy to address. However, I think the current algorithm / implementation might have additional limitations, which might make sense to have a look at as well while revising. For example, the definition of points similar to the other PR should also be added for rigid bodies.

bhosale2, 2022-08-01:
Ok, we are closing this for now and will revisit if this becomes relevant in the future.

closed 2022-08-01 by bhosale2