GazzolaLab/PyElastica #136: Removing _elastica_numba module

bhosale2, 2022-07-14:
Currently `numba` has become a default feature in `pyelastica` and the separate `_elastica_numba` module is deprecated since the past few version releases. The presence of that folder in future releases may create confusion as to if `numba` needs to be turned on or off in the simulator. As such, after discussion with @armantekinalp, we feel it may be time to remove that module altogether.

skim0119, 2022-07-15:
I remember removing `_elastica_numba` folder before and replaced with `_elastica_numba.py` for `v0.3.0`. I think the folder still exists because some past rebasing skipped removing `boundary_condition.py` file. Feel free to remove the folder and make a patch directly.

We still have `_elastica_numba.py` for user notification. In `v0.2`, we left all the files and threw warnings. These warnings will be changed to an error in `v0.3`. (Same story for `_elastica_numpy.py`) The purpose is to throw `ImportError` with our own message saying the user can directly import modules without specifying `_elastica_numba`. I think this message is needed because we have quite a bit of legacy code scattered around.

bhosale2, 2022-07-15:
Closed via #138.

PR #138 (2022-07-15): Remove _elastica_numba folder while keeping _elastica_numba.py
Fixes #136.
closed 2022-07-15 by bhosale2