GazzolaLab/PyElastica #142: Wrong docstring in `FixedConstraint` class

mstoelzle, 2022-07-17:
**Describe the bug**
The docstrings of the `nb_constraint_rotational_values`, `nb_constrain_translational_values` methods are faulty. The documentation mentions that both the `position_collection` and `fixed_position_collection` are in the shape `(dim, blocksize)`. But if you look at the code, `fixed_position_collection` is (probably) in the shape `(blocksize , dim)`: 
```py
@staticmethod
    @njit(cache=True)
    def nb_constraint_rotational_values(
        director_collection, fixed_director_collection, indices
    ) -> None:
        """
        Computes constrain values in numba njit decorator
        Parameters
        ----------
        director_collection : numpy.ndarray
            3D (dim, dim, blocksize) array containing data with `float` type.
        fixed_director_collection : numpy.ndarray
            3D (dim, dim, blocksize) array containing data with `float` type.
        indices : numpy.ndarray
            1D array containing the index of constraining nodes
        """
        block_size = indices.size
        for i in range(block_size):
            k = indices[i]
            director_collection[..., k] = fixed_director_collection[i, ...]

    @staticmethod
    @njit(cache=True)
    def nb_constrain_translational_values(
        position_collection, fixed_position_collection, indices
    ) -> None:
        """
        Computes constrain values in numba njit decorator
        Parameters
        ----------
        position_collection : numpy.ndarray
            2D (dim, blocksize) array containing data with `float` type.
        fixed_position_collection : numpy.ndarray
            2D (dim, blocksize) array containing data with `float` type.
        indices : numpy.ndarray
            1D array containing the index of constraining nodes
        """
        block_size = indices.size
        for i in range(block_size):
            k = indices[i]
            position_collection[0, k] = fixed_position_collection[i, 0]
            position_collection[1, k] = fixed_position_collection[i, 1]
            position_collection[2, k] = fixed_position_collection[i, 2]
```
The same issue exists analogue for `nb_constraint_rotational_values `

**To Reproduce**
```bash
python3 -m pytest -s
```

**Environment**
- PyElastica version: 0.3.0

bhosale2, 2022-07-19:
@mstoelzle good catch, do you want to fix this? should be a quick one (fixing function and corresponding tests)

mstoelzle, 2022-07-19:
> @mstoelzle good catch, do you want to fix this? should be a quick one (fixing function and corresponding tests)

@bhosale2 Can I integrate the fix it into PR #143?

bhosale2, 2022-07-19:
Hi, @mstoelzle I would request doing it in a separate PR. This keeps the issue-PR mapping one-to-one (easier to search later) and makes the PR to the point, shorter, and easier to review for us (and also easier for you to follow up on suggested changes).

skim0119, 2022-07-20:
resolved with #142

commit 8b120cc1 (2022-07-20): Fix #142 by adjusting data structure of `fixed_positions` and `fixed_directors` (#147)

* Fix #142 by changing order of `fixed_positions` and `fixed_directors`
* Remove empty initialization of `self.fixed_positions` and `self.fixed_directors`
closed 2022-07-20 by skim0119