GazzolaLab/PyElastica #153: `make install` changes lockfile, which is not ignored by git

mstoelzle, 2022-07-22:
I am not sure if this is expected behaviour, or a bug,  but I have the following question about the contributing workflow:

- I am running on the `update-0.3.0` branch
- As documented in the [`CONTRIBUTING.md`](https://github.com/GazzolaLab/PyElastica/blob/update-0.3.0/CONTRIBUTING.md) file, I run `make install`
- This will update the `poetry.lock` file. However, `poetry.lock` is not ignored. by git and appears as a change in my git working tree. I assume that I shouldn't commit this file, right?

bhosale2, 2022-07-22:
Hi @mstoelzle thanks for catching this, so per Poetry's version control system (https://python-poetry.org/docs/basic-usage/#commit-your-poetrylock-file-to-version-control), if there exists a `.lock` file one should only call `poetry install` to install the correct version locked dependencies. After that, if the developer feels new dependencies need to be added, then the `.toml` file should be modified, followed by running the current version of `make install`, which locks the versions and then install (only then `.lock` commit is needed)

For now please just run `poetry install` to install dependencies and not change the `.lock` file. We will soon do the fix for this from our side.

bhosale2, 2022-07-22:
Fixed by #155.

PR #155 (2022-07-22): (#153) fix poetry install instructions
Fixes #153.
closed 2022-07-22 by bhosale2