GazzolaLab/PyElastica #159: Illustration of cylinders is incorrect

matei1996, 2022-07-28:
Dear PyElastica Team,

since yesterday my illustration of my experiment changed. I am trying to guide a rod through labyrinth of several cylinders. However, it seems like the basis of cylinders is not circular anymore. It looks more like an ellipsis. I updated my Anaconda(now 4.13.0) and Jupyter Notebook(now 6.4.11) yesterday. I am using a Python3(ipykernel). Maybe this causes the issue. I have attached two example videos. One is older with circular basis, one is new where my described problem can be seen.

Kind regards,
Matei

https://user-images.githubusercontent.com/87129296/181478400-8ad75ce4-437c-42f0-853a-4ad567dd7165.mp4

https://user-images.githubusercontent.com/87129296/181478516-60dc7bdf-5bd0-43ee-96c2-72e9f487d86d.mp4

bhosale2, 2022-07-28:
@matei1996 can you please post your case file code, the function you use to plot the cylinders, and the code for that function?

armantekinalp, 2022-07-28:
Also please list the conda environment variables and their version.

matei1996, 2022-07-28:
The code is below In the first part the function to plot and create videos is defined. The second part generates the rod and cylinders. It is a bit messy, sorry about that

The pdf 
[Conda environment variables.pdf](https://github.com/GazzolaLab/PyElastica/files/9212574/Conda.environment.variables.pdf)
file contains the variables and version (I hope this is what you are asking for)

`
%reset -f
import numpy as np
from matplotlib import pyplot as plt
import cv2
import os

def make_data_for_cylinder_along_y(cstart, cradius, cheight):
    center_x, center_z = cstart[0], cstart[1]
    y = np.linspace(0, cheight, 5)
    theta = np.linspace(0, 2 * np.pi, 20)
    theta_grid, y_grid = np.meshgrid(theta, y)
    x_grid = cradius * np.cos(theta_grid) + center_x
    z_grid = cradius * np.sin(theta_grid) + center_z
    y_grid += cstart[2]
    return [x_grid, y_grid, z_grid]

def plot_video(
    rod_history: dict,
    cylinder_history: dict,
    video_name="video.mp4",
    margin=0.2,
    fps=60,
    step=1,
    *args,
    **kwargs
):  # (time step, x/y/z, node)
    c=-1
    num_cyl=len(cylinder_history)-1
    for item in cylinder_history:
        c=c+1
        #cylinder_radius=cylinder_radius[c]
        #current_cylinder=cylinder_history[c]
        #breakpoint()
        cylinder_start = np.array(item["position"])[0, ...]
        cylinder_radius = kwargs.get("cylinder_radius")[c]
        cylinder_height = kwargs.get("cylinder_height")
        cylinder_direction = kwargs.get("cylinder_direction")

        XC, YC, ZC = make_data_for_cylinder_along_y(
            cylinder_start, cylinder_radius, cylinder_height
        )

        import matplotlib.animation as manimation

        plt.rcParams.update({"font.size": 22})

        # Should give a (n_time, 3, n_elem) array
        positions = np.array(rod_history["position"])
        # (n_time, 3) array
        com = np.array(rod_history["com"])

        cylinder_com = np.array(item["com"])
        cylinder_origin = cylinder_com - 0.5 * cylinder_height * cylinder_direction

        print("plot video")
        FFMpegWriter = manimation.writers["ffmpeg"]
        metadata = dict(title="Movie Test", artist="Matplotlib", comment="Movie support!")
        writer = FFMpegWriter(fps=fps, metadata=metadata)
        dpi = 50

        # min_limits = np.roll(np.array([0.0, -0.5 * cylinder_height, 0.0]), _roll_key)
        from mpl_toolkits.mplot3d import Axes3D

[... 293 lines cut ...]
cylinder_radius_test=[cylinder_radius,cylinder_radius_2,cylinder_radius_3]

integrate(timestepper, single_rod_sim, final_time, total_steps)

if PLOT_FIGURE:
    plot_video(
        recorded_rod_history,
        #recorded_cyl_history_2,
        all_cylinders,
        "cylinder_rod_collision.mp4",
        cylinder_direction=cylinder_direction,
        cylinder_height=cylinder_height,
        cylinder_radius=cylinder_radius_test,
    )

    plot_cylinder_rod_position(
        recorded_rod_history,
        #recorded_cyl_history_2,
        all_cylinders,
        cylinder_radius=cylinder_radius_test,
        rod_base_radius=base_radius,
        TIP_COLLISION=TIP_COLLISION,
        TIP_CHOICE=TIP_CHOICE,
    )
`

skim0119, 2022-07-28:
@matei1996 I briefly looked, and I don't see any issue in the code other than figsize is not square. I would say try to set the axis to be the equal length in both x and y (set aspect ratio to be 1), and leave an issue to Matplotlib if the problem remains.

matei1996, 2022-07-29:
Setting the aspect ratio to 1 solved the problem. Thanks for the help! I am closing the issue.

closed 2022-07-29 by matei1996