GazzolaLab/PyElastica #169: Update visualisation example to use ffmpeg instead of moviepy

bhosale2, 2022-08-06:
Currently, the continuum snake example in `examples/Visualisation` uses `moveipy` for generating videos, unlike other example cases where we use `ffmpeg` throughout. To maintain consistency, we need to update the example mentioned above to use `ffmpeg` instead of `moviepy`.

Shikharishere, 2022-10-20:
Hey @bhosale2 is this issue still open. If yes, I am happy to contribute...🙂

Rahul-JOON, 2023-02-20:
If this issue is still open, I would like to be assigned to it. @armantekinalp @bhosale2

armantekinalp, 2023-02-20:
Yes this issue is open.

Rahul-JOON, 2023-02-21:
Hi @armantekinalp, Firstly thank you for responding and assigning me this issue. But I ran into a problem and would like your help.

Before I could implement the solution to the issue, I went ahead and set up my working environment as per [Contributing guidelines](https://github.com/GazzolaLab/PyElastica/blob/master/CONTRIBUTING.md); and resolved all the dependencies for examples too.

However, when I ran the examples/visualization bash code, the below exception was thrown on the second step
[image]

Tracing back I noticed "tid" was not being recognized by Pylance in VS Code:
[image]

I tried installing tid manually too, but it returns :
`(pyelastica-dev) C:\Windows\system32>pip install tid
Collecting tid
  Using cached tid-0.2.0.tar.gz (2.2 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: tid
  Building wheel for tid (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for tid (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [38 lines of output]
      A setup.py file already exists. Using it.
      Compiling tid/_gettid.pyx because it changed.
      [1/1] Cythonizing tid/_gettid.pyx
      running build
      running build_py
      creating C:\Users\RAHUL\AppData\Local\Temp\pip-install-q9mfnc7g\tid_8874eb78fe9d4d99a363a85c66bf1149\build
      creating C:\Users\RAHUL\AppData\Local\Temp\pip-install-q9mfnc7g\tid_8874eb78fe9d4d99a363a85c66bf1149\build\lib.win-amd64-3.8
      creating C:\Users\RAHUL\AppData\Local\Temp\pip-install-q9mfnc7g\tid_8874eb78fe9d4d99a363a85c66bf1149\build\lib.win-amd64-3.8\tid
      copying tid\__init__.py -> C:\Users\RAHUL\AppData\Local\Temp\pip-install-q9mfnc7g\tid_8874eb78fe9d4d99a363a85c66bf1149\build\lib.win-amd64-3.8\tid
      copying tid\_gettid.c -> C:\Users\RAHUL\AppData\Local\Temp\pip-install-q9mfnc7g\tid_8874eb78fe9d4d99a363a85c66bf1149\build\lib.win-amd64-3.8\tid
      copying tid\_gettid.pyx -> C:\Users\RAHUL\AppData\Local\Temp\pip-install-q9mfnc7g\tid_8874eb78fe9d4d99a363a85c66bf1149\build\lib.win-amd64-3.8\tid
      running build_ext
      building 'tid._gettid' extension
      C:\Users\RAHUL\AppData\Local\Temp\pip-build-env-e1dabl_t\overlay\Lib\site-packages\Cython\Compiler\Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: C:\Users\RAHUL\AppData\Local\Temp\pip-install-q9mfnc7g\tid_8874eb78fe9d4d99a363a85c66bf1149\tid\_gettid.pyx
        tree = Parsing.p_module(s, pxd, full_module_name)
      C:\Users\RAHUL\anaconda3\envs\pyelastica-dev\lib\distutils\dist.py:274: UserWarning: Unknown distribution option: 'python_requires'
        warnings.warn(msg)
      error: Unable to find vcvarsall.bat
      Traceback (most recent call last):
        File "C:\Users\RAHUL\anaconda3\envs\pyelastica-dev\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 351, in <module>
          main()
        File "C:\Users\RAHUL\anaconda3\envs\pyelastica-dev\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 333, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "C:\Users\RAHUL\anaconda3\envs\pyelastica-dev\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 249, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
        File "C:\Users\RAHUL\AppData\Local\Temp\pip-build-env-e1dabl_t\overlay\Lib\site-packages\poetry\core\masonry\api.py", line 56, in build_wheel
          return WheelBuilder.make_in(
        File "C:\Users\RAHUL\AppData\Local\Temp\pip-build-env-e1dabl_t\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 85, in make_in
          wb.build(target_dir=directory)
        File "C:\Users\RAHUL\AppData\Local\Temp\pip-build-env-e1dabl_t\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 118, in build
          self._build(zip_file)
        File "C:\Users\RAHUL\AppData\Local\Temp\pip-build-env-e1dabl_t\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 179, in _build
          self._run_build_command(setup)
        File "C:\Users\RAHUL\AppData\Local\Temp\pip-build-env-e1dabl_t\overlay\Lib\site-packages\poetry\core\masonry\builders\wheel.py", line 217, in _run_build_command
          subprocess.check_call(
        File "C:\Users\RAHUL\anaconda3\envs\pyelastica-dev\lib\subprocess.py", line 364, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['C:/Users/RAHUL/anaconda3/envs/pyelastica-dev/python.exe', 'C:\\Users\\RAHUL\\AppData\\Local\\Temp\\pip-install-q9mfnc7g\\tid_8874eb78fe9d4d99a363a85c66bf1149\\setup.py', 'build', '-b', 'C:\\Users\\RAHUL\\AppData\\Local\\Temp\\pip-install-q9mfnc7g\\tid_8874eb78fe9d4d99a363a85c66bf1149\\build']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for tid
Failed to build tid
ERROR: Could not build wheels for tid, which is required to install pyproject.toml-based projects

(pyelastica-dev) C:\Windows\system32>`

I would really appreciate if you could help me with this.
I am using python-3.8.0

Thanks in Advance!

Rahul-JOON, 2023-02-24:
Hi @armantekinalp, I have created a pull request to solve this issue, please check it at [#169](https://github.com/GazzolaLab/PyElastica/pull/232)

bhosale2, 2023-02-24:
Closed via #232.

closed 2023-02-24 by bhosale2