GazzolaLab/PyElastica #176: Release v0.3 Checklist

skim0119, 2022-08-16:
# Release Checklist

- [ ] Update RELEASE.md - It should be in based on milestone description.
- [ ] Commit the changes: 
```bash
git add RELEASE.md
git commit -m "Changelog for upcoming release <version>"
```
- [ ] Final check `README` and `CONTRIBUTING`.
- [ ] Bump LICENSE year
- [ ] Bump version
- [ ] Dependency version (`poetry update`)
  - Might break some dependency chain.

### Test
- [ ] Run the tests
- [ ] Run the formatting
- [ ] CI Check
```
python -m pytest
make all
```

### Setup.py
- [ ] Update version number in setup.py (or version.py) (can also be minor or major)
- [ ] Check if we need to upgrade `REQUIRES_PYTHON` or `REQUIRED` dependencies.
- [ ] Check if local installation can be done.
- [ ] Update `download_url` in setup
```bash
conda create --name temp
conda activate temp
conda install python==<version>
which pip
pip install .
```

### Documentation and Website
- [ ] Check all documentations are updated
- [ ] Update version label on documentation
  - [ ] docs/conf.py
  - [ ] Python version
- [ ] Update version label on website
  - [ ] Python version

```diff
- BEYOND THIS, EACH STEP SHOULD BE DONE IN SERIAL (STEP_BY_STEPS). -
```

### Last Edit
- [ ] Rebase the commit history (should be done by only one person) right before the PR. (Not required)

### Update
1. [ ] Merge the branch to `master`
2. [ ] Release on PyPI by uploading `python setup.py upload` (make sure twine is installed)
3. [ ] Release on github (v<version>) "include v in front"

### Final Check
- [ ] Test that it pip installs
- [ ] Check the PyPI listing page to make sure that the README, release notes, and roadmap display properly.
- [ ] Edit the release-page on GitHub. Paste the release notes with a title for the release.
- [ ] Check Codcov ID and make sure they are running.
- [ ] Make mirror branch in private repo.

armantekinalp, 2022-08-16:
I think all codes for this update merged. So I will start testing examples.

armantekinalp, 2022-08-18:
Example cases are good and running

PR #180 (2022-08-18): Update 0.3.0
Fixes #176.
closed 2022-08-18 by skim0119