GazzolaLab/PyElastica #178: 🔥 Bump up python version for ReadTheDocs

skim0119, 2022-08-17:
**Describe the bug**
Within RTD, documentation is built using python 3.7, and the build is failing.

**Details**
We are slowly adopting the `typing` and `Protocol` module, which is included in version 3.8+. Previously, `Protocol` was in `typing_extensions`.

**What to do**
Modify `.readthedocs.yaml` to install package using poetry.

**Reference**
https://github.com/GazzolaLab/MiV-OS/blob/main/.readthedocs.yaml

skim0119, 2022-08-17:
[image]

armantekinalp, 2022-08-17:
fixed with [6c8e518](https://github.com/GazzolaLab/PyElastica/commit/6c8e518e24b6bccfc9be5917738aa7de95716a4a)

closed 2022-08-17 by armantekinalp