GazzolaLab/PyElastica #193: Clarify namespace for elastica modules to avoid namespace confusion and clutter

bhosale2, 2022-08-30:
Currently, in example cases, the modules from `elastica` are imported directly as `from elastica import *`. This can become an issue when using `elastica` across packages (where `elastica` is a plugin), where modules from other packages can collide in namings with the modules from `elastica`. To avoid this, it may be helpful to use functionalities in `elastica` as:

`import elastica as ea; ea.Damping(....)` to make it explicit where the functionality comes from.

@skim0119 and @armantekinalp thoughts?

armantekinalp, 2022-08-30:
I think it is a good suggestion.

skim0119, 2022-08-31:
Last point on [pep8](https://peps.python.org/pep-0008/#imports)

prabhashj07, 2023-02-08:
Is this issue open?
I'm a beginner and I'd like to work on it.

armantekinalp, 2023-02-08:
Yes it is still open

Rahul-JOON, 2023-02-18:
If this issue is still open, I would like to be assigned to it.

Deep-Rai-08, 2023-03-13:
Is this issue is open?? @armantekinalp

skim0119, 2023-03-14:
@Deep-Rai-08 The issue is still open. Feel free to make a PR.

bhosale2, 2023-03-15:
A note to everyone interested in this issue: if you plan to work on it please either assign yourself to the issue or let us know that you wish to be assigned to the issue.

amangupta201, 2023-03-20:
NameSpaces can be a great way to sort and clarify the elastica modules
Please assign me to this issue

AsadNizami, 2023-03-27:
Hi, is anyone still working on the issue? If not I'd like to work on it.

bhosale2, 2023-04-25:
Closed via #238.

closed 2023-04-25 by bhosale2