GazzolaLab/PyElastica #194: Remove the option for internal damping module for Cosserat rods following warning message

bhosale2, 2022-08-31:
Currently, adding damping to the rod via `AnalyticalLinearDamper` results in the following warning message:

[image]

Per the warning to make a soft change, we should convert the warning message to an error message if the user provides internal damping other than `None` when initializing a Cosserat rod, and remove related functionality from the cosserat rod equations.

Thoughts @armantekinalp and @skim0119?

armantekinalp, 2022-08-31:
I agree with you suggestion. As we discussed with @bhosale2, I think we should remove the internal damping functionality, all variables and tests as we planned in `v.0.3.1` and update the warning message accordingly.

bhosale2, 2023-05-03:
Closed via #243.

PR #243 (2023-05-03): (#194) remove internal damping option for Cosserat rod
Fixes #194.
closed 2023-05-03 by bhosale2