GazzolaLab/PyElastica #204: ModuleNotFoundError__No Module named 'examples'

LegendCC-Zhang, 2022-10-15:
Dear all, I used anaconda as envs to pip install pyelastica, and then I tried to do  some examples given here.

But here some issues occurred, ModuleNotFoundError: No module named "examples". And then I found that we need do “make install_examples_dependencies” first, but I do not know how to do it.
[image]
[image]

LegendCC-Zhang, 2022-10-15:
Looking forward to your help! Thank you in advance!

bhosale2, 2022-10-15:
@LegendCC-Zhang can you please provide more information on this issue? For instance which example did you try to run? Did you run the python example script or the jupyter notebook example?

LegendCC-Zhang, 2022-10-16:
> 
Thanks for your reply! 

I used anaconda to create an env.
[image]
, then "pip install pyelastica", and then used jupyter notebook to run the examples. 
For the examples, such as "[timoshenko_postprocessing.py](https://github.com/GazzolaLab/PyElastica/blob/master/examples/TimoshenkoBeamCase/timoshenko_postprocessing.py)" that require import example module, error occurs at my end. 
I have two laptops, the OS of which are Windows and Linux, respectively. But I failed to do the "make install_examples_dependencies", as stated in the readme shown in the examples file (https://github.com/GazzolaLab/PyElastica/tree/master/examples).
The attached file is the pip list based on Windows-anaconda-pyelastica  env.
So, could you please help on this issue?

Thanks!

armantekinalp, 2022-10-16:
@LegendCC-Zhang can you follow these steps

1. Please clone the code and `cd` inside the cloned directory
2. Do `make poetry-download`
3. Do `make install`
4. Do `make install_example_dependencies`
5. `pip install pyelastica`
6. Then please run `python examples/TimoshenkoBeamCase/timoshenko.py `

Let me know if you have any problems.

closed 2022-10-19 by armantekinalp