GazzolaLab/PyElastica #211: Endpoint force application

danielesmm, 2022-11-07:
**Describe the setup**
Hi all,
my idea is to have a single rod, apply a force at the endpoint, measure the force that is applied (expoliting callbacks) and then apply an opposite force to counteract the external load application. I want the rod to keep the straight position even if forces are applied on it. Do you know how to apply the counteracting force once the simulation started? Thank you

armantekinalp, 2022-11-07:
Hi @danielesmm ,

You can add two  [EndpointForces](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/external_forces.py#L128) with a force vector opposite direction to the simulator. 

```python
    end_force = np.array([0, 1, 0])

    simulator.add_forcing_to(rod).using(
        EndpointForces, 0.0 * end_force, end_force, ramp_up_time=final_time / 2.0
    )
    simulator.add_forcing_to(rod).using(
        EndpointForces, 0.0 * end_force, -end_force, ramp_up_time=final_time / 2.0
    )
```

armantekinalp, 2022-11-08:
Hi @danielesmm 

Could you please share the error here? From your description, I think you need to write a new forcing class that dynamically adjust the applied force on the rod. Basically you get the external forces applied using callback, store it in some array and use that array again in your new forcing class to apply on the rod.