GazzolaLab/PyElastica #215: Rigid connection between two rods

danielesmm, 2022-11-23:
Hi all,
I need help for my project..
I have two rods directed along the x direction that are parallel to each other. The two rods are connected by means of a third rod along the y direction. Do you how to fix the connecting rod to the two rods without using joints? 
I tried with joints but the connection is not strong enough. 
I want to use a sort of boundary condition so that the two ends of the horizontal rod are move together with the vertical ones. My goal is to apply a force/torque to one of the two vertical rod and moving the second accordingly so that they will be always parallel to each other.
Thank you for your help,
Daniele

armantekinalp, 2022-11-24:
Hi @danielesmm ,

I think joints is the best way to do these type of connections. But if you want to implement a boundary condition, you can set the node position of connecting nodes to each other. This is a hacky way to do it.

You need to define a new BC class which takes rod one and rod two and their connection indexes. Make rod one rod two and indexes as members of this class. Then at the apply constrain stage just set node position of rod one to the other.

danielesmm, 2022-11-24:
Thank you @armantekinalp ,
I think that defining the BC class is the best way. However, I have no idea on how constrain the two nodes.

armantekinalp, 2022-11-24:
Implementation wise I suggest reviewing this implementation first https://github.com/GazzolaLab/PyElastica/blob/master/elastica/boundary_conditions.py#L486

You need to update the class input arguments and take rod_one and rod_two as well and make them references of your new bc class. Give it a shot and share the code here if you have questions.

danielesmm, 2022-11-24:
class ConstrainTwoRods(GeneralConstraint):

    def __init__(self,
                rod2,
                index1,
                index2,
                *args,
                **kwargs):

        super().__init__(
            *args,
            translational_constraint_selector=np.array([True, True, True]),
            rotational_constraint_selector=np.array([True, True, True]),
            **kwargs,
        )
        self.rod2 = rod2
        self.index1 = index1
        self.index2 = index2

    def constrain_values(
        self, rod: Union[Type[RodBase], Type[RigidBodyBase]], time: float
    ) -> None:
            self.nb_constrain_translational_values(self,
                rod.position_collection,
            )
            self.nb_constraint_rotational_values(
                rod.director_collection,
            )

    def constrain_rates(
        self, rod: Union[Type[RodBase], Type[RigidBodyBase]], time: float
    ) -> None:
            self.nb_constrain_translational_rates(
                rod.velocity_collection,
            )
            self.nb_constrain_rotational_rates(
                rod.omega_collection,

            )

    @staticmethod
    @njit(cache=True)
    def nb_constraint_rotational_values(self,
        director_collection
    ) -> None:

        director_collection[...,self.index1] = self.rod2.director_collection[...,self.index2]

    @staticmethod
    @njit(cache=True)

    def nb_constrain_translational_values(self,
        position_collection
    ) -> None:

        position_collection[...,self.index1] = self.rod2.position_collection[...,self.index2]

    @staticmethod
    @njit(cache=True)

    def nb_constrain_translational_rates(self,velocity_collection) -> None:

        velocity_collection[0, self.index1] = self.rod2.velocity_collection[0, self.index2]
        velocity_collection[1, self.index1] = self.rod2.velocity_collection[1, self.index2]
        velocity_collection[2, self.index1] = self.rod2.velocity_collection[2, self.index2]

    @staticmethod
    @njit(cache=True)
    def nb_constrain_rotational_rates(self,omega_collection) -> None:

        omega_collection[0, self.index1] = self.rod2.omega_collection[0, self.index2]
        omega_collection[1, self.index1] = self.rod2.omega_collection[1, self.index2]
        omega_collection[2, self.index1] = self.rod2.omega_collection[2, self.index2]

This is the code but I have this error:
numba.core.errors.TypingError: Failed in nopython mode pipeline (step: nopython frontend)
non-precise type pyobject
During: typing of argument at C:\Users\danie\AppData\Local\Programs\Python\Python310\lib\site-packages\elastica\boundary_conditions.py (831)

File "..\..\..\..\AppData\Local\Programs\Python\Python310\lib\site-packages\elastica\boundary_conditions.py", line 831:
    def nb_constrain_translational_values(self,
        <source elided>

        position_collection[...,self.index1] = self.rod2.position_collection[...,self.index2]
        ^

This error may have been caused by the following argument(s):
- argument 0: Cannot determine Numba type of <class 'elastica.boundary_conditions.ConstrainTwoRods'>

Do you know what's wrong with my code?

armantekinalp, 2022-11-24:
Yes, remove the `njit` decorators. Numba cannot access to the class self variables.

closed 2022-11-28 by danielesmm