GazzolaLab/PyElastica #216: "install_examples_dependencies" file is missing

saultshaker, 2022-12-04:
Hi, 

I noticed in the documentation in the README for the [examples folder](https://github.com/GazzolaLab/PyElastica/tree/master/examples), under the "Installing Requirements" section, there is mention of a "install_examples_dependencies"  to be run with "make install_examples_dependencies".  

However, I can't seem to find a file called  "install_examples_dependencies" anywhere. 

Thanks for your time!

armantekinalp, 2022-12-04:
Hi @saultshaker ,

You need to run `make install_examples_dependencies` dependencies at the `PyElastica` folder where `Makefile` is located and not inside `PyElastica/examples/`

saultshaker, 2022-12-04:
Thank you! This resolved my issue :)

bhosale2, 2022-12-27:
Closing since resolved.

closed 2022-12-27 by bhosale2