GazzolaLab/PyElastica #222: time dependent parameters

fepauly, 2023-01-04:
Hey,

for some future projects i need to adjust some parameters of the rod during simulation depending on time, pressure or length. I was thinking about writing a callback, but changes on the system like the radius (for testing purposes) doesn't seem to affect the simulation. Hence, my question: Is it possible to change parameters of the rod during simulation depending on for example time?
And additionally, is pressure actuation possible or considered for future updates?

Thank you very much for your time and feedback.

Bests regards,
Felix

skim0119, 2023-01-04:
Hi @fepauly,

> Is it possible to change parameters of the rod during simulation depending on for example time?

We don't recommend changing the rod parameters in the middle of the simulation since it might create potential instability. Still, you should be able to change the parameters of the rod by directly accessing the variables, `rod.<variable name>`. The problem with `radius` particularly is that it is "computed" variable at each step, which is why changing the `radius` wouldn't affect the simulation.

Before computing Cosserat rod equation, the radius is pre-computed to conserve the volume. (https://github.com/GazzolaLab/PyElastica/blob/cab9ea9b95a3b6b7075618dc4477bd85d5fe5f5d/elastica/rod/cosserat_rod.py#L548) If you want to have the effect of changing `radius`, consider changing `volume` instead.

I believe this can be done within the `Callback`. If you have trouble implementing it, please leave a comment.

> is pressure actuation possible or considered for future updates?

It is possible but we are not planning to add it as a native feature at this stage, because the constitutive relation for pressure versus actuation varies case by case. We typically achieve this by first empirically map the pressure vs force/moment, and implement them as uniform force/moment in simulation.

skim0119, 2023-01-26:
I'm closing this issue. Please re-open if you have further question.

closed 2023-01-26 by skim0119