GazzolaLab/PyElastica #233: Fix the docs.

armantekinalp, 2023-02-24:
**Describe the bug**
Docs are failing.

**To Reproduce**
Please see the [build](https://github.com/GazzolaLab/PyElastica/actions/runs/4264369787/jobs/7422359766)

bhosale2, 2023-02-24:
We also need to update the build system and scripts for docs. This could be a good reference for generating modern doc-building system:

https://github.com/cjolowicz/cookiecutter-hypermodern-python/blob/main/.readthedocs.yml

armantekinalp, 2023-02-24:
@bhosale2 suggested this repo as well https://github.com/tp5uiuc/elastica-pipelines

bhosale2, 2023-03-15:
An update here, @skim0119 has removed the failing Github action test which is probably redundant at this time. This is because first, the repo for that seems outdated, and second, the docs are rendering without any failures. We can modify this as an issue that updates the docs building scripts and thus is low priority.

armantekinalp, 2023-12-05:
1772679

closed 2023-12-05 by armantekinalp