GazzolaLab/PyElastica #236: Can closed chains of rods be constructed?

QuantuMope, 2023-03-16:
Hello,

I was wondering if Elastica is capable of simulating closed chain loops of rods.
For example, if I were to have four rods, each with a joint as indicated by J as shown below

        rod 1
J ---------------------- J
|                            |
|rod2                    |rod3
J -----------------------J
      rod 4
      
Does the Elastica framework allow for such closed chain configurations?
I wasn't able to see any examples of this in the documentation.

Thanks in advance!

armantekinalp, 2023-03-16:
Hi @QuantuMope,

You can connect rods using [FreeJoint](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/joint.py#L10) class one by one to each other ([Here](https://github.com/GazzolaLab/PyElastica/blob/master/examples/JointCases/spherical_joint.py) is an example simulation you can see usage ) . But this Freejoint only connect the end nodes of rods and does not enforce orientation. For example if you want rods to be perpendicular to each other you need to write a new joint class. You can check existing joint classes [in this file](https://github.com/GazzolaLab/PyElastica/blob/master/elastica/joint.py)

QuantuMope, 2023-03-16:
Hi @armantekinalp,

Thank you for your prompt response. I just ran the examples in spherical_joint.py but this still only shows functionality for a non-chained configuration of rods. I'm not necessarily interested in maintaining perpendicularity of rods rather I would like to know if Elastica is capable of creating closed chained loops.

For example, the drawing in my previous post shows a square structure whose sides are all elastic rods.

bhosale2, 2023-05-08:
Closing due to inactivity.

closed 2023-05-08 by bhosale2