GazzolaLab/PyElastica #270: Mesh Object Class

Ali-7800, 2023-06-19:
## Mesh Object Class Requirements
### Initialization
- [ ] Can be initialized by reading a STL or OBJ mesh file formats.

### Class Attributes
- [ ] faces, dimension: (3 spatial coordinates, 3 vertices, n faces)
This should store the coordinates of the 3 vertices of each of the n faces of the imported mesh.

- [ ] face normals, dimension: (3 spatial coordinates, n faces)
This should store the coordinates of the unit normal vector of each of the n faces.

- [ ] face centers,  dimension: (3 spatial coordinates, n faces)
This should store the coordinates of the position vector of each of the n face centers (average of the vertices).

- [ ] mesh orientation, dimension: (3 spatial coordinates, 3 vectors)
This should store the 3 orthonormal basis vectors that define the mesh orientation, by default this should be the cartesian orthonormal basis vectors.

- [ ] mesh scale, dimension (3 lengths,)
This should store the 3 dimensions of the smallest box that could fit the mesh whose sides align with the mesh orientation vectors.

- [ ] mesh center,  dimension: (3 spatial coordinates,)
This should store the coordinates of the position vector of the center of the smallest box that could fit the mesh.

### Class Methods
- [ ] Method for visualizing the mesh object
- [ ] Method for translating the mesh
Given a position vector; translate the faces, face centers, and mesh center.
- [ ] Method for reorienting the mesh
Given a rotation axis (unit vector) and an angle; rotate the faces, face centers, face normals, and mesh center.
- [ ] Method for rescaling the mesh
Given a scale value; scale the faces, face centers, and mesh scale.

Ali-7800, 2023-06-19:
@Rahul-JOON can you work on this issue

Rahul-JOON, 2023-06-20:
Sure, I will take this on.