GazzolaLab/PyElastica #276: Error in generating video with continuum snake example

AkashVardhan7, 2023-07-06:
Hello I am newbie with using Elastica and I installed everything (the examples and the docs) as mentioned on your website. I was able to get the first axial stretching example to work fine, but then I moved to try the continuum snake example, as I will be intend to simulate worm locomotion, and am running into the problem of writing the video as ffmpeg isn't being recognized. I will copy paste the error message down below: 

File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\continuum_snake_postprocessing.py", line 65, in plot_video
    FFMpegWriter = manimation.writers["ffmpeg"]
  File "C:\Users\avardhan7\.conda\envs\pyElastica\lib\site-packages\matplotlib\animation.py", line 148, in __getitem__
    raise RuntimeError(f"Requested MovieWriter ({name}) not available")
RuntimeError: Requested MovieWriter (ffmpeg) not available

I made sure to install python-ffmpeg, and just the ffmpeg packages and have it installed in my pyElastica environment, yet can't understand why this issue is coming up. The code is able to generate the plots of the velocity and curvatures, but crashes at the video generation.

Rahul-JOON, 2023-07-09:
Hi @AkashVardhan7, Can you check if FFmpeg is installed correctly in the environment by running 'ffmpeg -version', if ffmpeg is installed correctly/added to the path, this should return some information about the version of ffmpeg; if this results in an error, I suggest you to use conda to install ffmpeg in your environment by "conda install ffmpeg", then repeating the verifying process again.

Moreover please also check if you have installed POV-ray in your system.
Bonus note: Turn on auto close povray for winodows after rendering to avoid spamming hundreds of povray windows on your system while rendering.

Rahul-JOON, 2023-07-11:
Hi @AkashVardhan7, Why did you change the name of the variable? Though The error states otherwise, the name discrepancy can lead to another NameError.

Moreover, your error seems to be solely based on environmental setup (ffmpeg installation to be exact) and not the code, this could be occurring if ffmpeg is not configured correctly in pycharm environments; Try troubleshooting this in pycharm.

If nothing works, I suggest you to create a virtual env using anaconda and executing the code via windows powershell, if you still prefer to use coding IDE, you can try using VSCode and setting up anaconda envs there.

AkashVardhan7, 2023-07-11:
I had left it unchanged initially and was still encountering the same error and so decided to debug by following the suggestion from the IDE, here's the error message, I will try setting it up in VscCode and/or debugging with pycharm..

Traceback (most recent call last):
  File "C:\Users\avardhan7\AppData\Local\JetBrains\PyCharm 2023.1.2\plugins\python\helpers\pydev\pydevconsole.py", line 364, in runcode
    coro = func()
  File "<input>", line 1, in <module>
  File "C:\Users\avardhan7\AppData\Local\JetBrains\PyCharm 2023.1.2\plugins\python\helpers\pydev\_pydev_bundle\pydev_umd.py", line 198, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Users\avardhan7\AppData\Local\JetBrains\PyCharm 2023.1.2\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\Continuum_Snake.py", line 241, in <module>
    [avg_forward, avg_lateral, pp_list] = run_snake(
  File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\Continuum_Snake.py", line 170, in run_snake
    plot_video(
  File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\continuum_snake_postprocessing.py", line 67, in plot_video
    FFMpegWriter = manimation.writers["ffmpeg"]
  File "C:\Users\avardhan7\.conda\envs\pyElastica\lib\site-packages\matplotlib\animation.py", line 148, in __getitem__
    raise RuntimeError(f"Requested MovieWriter ({name}) not available")
RuntimeError: Requested MovieWriter (ffmpeg) not available

Rahul-JOON, 2023-07-16:
Hi @AkashVardhan7, any update/progression?

AkashVardhan7, 2023-07-19:
I switched to VS CODE and the issue persists, ffmpeg isn't being detected by the code although I have verified that the libraries are properly installed in my environment.

# packages in environment at C:\Users\avardhan7\.conda\envs\pyElastica:
#
# Name                    Version                   Build  Channel
ffmpeg                    1.4                      pypi_0    pypi
imageio-ffmpeg            0.4.8              pyhd8ed1ab_0    conda-forge
python-ffmpeg             2.0.4                    pypi_0    pypi

Further I manually placed the executable of ffmpeg extracted from the website in the folder C:\Users\avardhan7\.conda\envs\pyElastica\Library\bin and created an enviroment variable with the same path in my computer.
[image]

Yet the error persists as you can see from
[image]
I am yet to try POV-RAY

AkashVardhan7, 2023-07-19:
I looked at the visualization code for POV ray it requires ffmpeg dependency to work properly for the video writer, so I guess it's not useful until I am able to resolve the reason for this discrepancy. Also, your notebooks on Binder are unable to install numpy for some reason and keep giving me the error ModuleNotFoundError: No module named 'numpy'

armantekinalp, 2023-07-19:
> Also, your notebooks on Binder are unable to install numpy for some reason and keep giving me the error ModuleNotFoundError: No module named 'numpy'

Thanks for letting us know, I created an issue #282  for this problem and we will work on it.

AkashVardhan7, 2023-07-23:
Got it to work with pillow. 
 PillowWriter = manimation.writers["pillow"]
  writer = PillowWriter(fps=fps, metadata=metadata)
and changing the extensions of the video files from .mp4 to .gif enables me to generate the video as a gif...

armantekinalp, 2023-07-24:
@AkashVardhan7 thank you for sharing how you resolved this issue. 

I am closing this issue, in the case more discussion needed please reopen.

closed 2023-07-24 by armantekinalp