GazzolaLab/PyElastica #282: Binder is not working with Poetry

armantekinalp, 2023-07-19:
**Describe the bug**
Binder examples for PyElastica does not run on the server, because MyBinder requires a `requirement.txt` file and PyElastica uses poetry as the dependency manager. [Here](https://mybinder.readthedocs.io/en/latest/examples/sample_repos.html) are the some example Binder repositories that might be useful to find a work around.

bhosale2, 2023-07-24:
@armantekinalp two suggestions maybe from my side:

1. Have a `requirements.txt` on the main repo and use that (many Python packages I saw keep multiple dependency manager files that include `requirements.txt`, `env.yaml` for conda, etc)
2. Call `poetry` at the start of notebook to export toml as `requirements.txt` file?

bhosale2, 2023-08-09:
@armantekinalp please assign the appropriate priority here

armantekinalp, 2023-12-05:
Maybe collar

Ali-7800, 2024-03-10:
Closing as issue is solved

closed 2024-03-10 by Ali-7800