GazzolaLab/PyElastica #285: Issue with Plectoneme_case example

AkashVardhan7, 2023-07-26:
Hi I was running the plectoneme_case example, and the code crashed after generating videos, at the compute_twist function. Here's a screenshot.
[image]

armantekinalp, 2023-07-28:
Can you share the list of packages you installed in your environment? I suspect this is something local in your settings. Two of us able to run this example without an issue.

armantekinalp, 2023-07-28:
We have resolved this issue, but haven't pushed to the main branch yet. It will be pushed in our next release cycle. However, you can still run the example.  Checkout `update-0.3.2` branch and run the example with that.

closed 2023-08-07 by armantekinalp