GazzolaLab/PyElastica #288: Rendering videos with RL cases

AkashVardhan7, 2023-08-07:
I am running elastica on Windows and intend to create a project of a worm crawling across heterogenous environments and then implement Reinforcement learning to optimize some aspect of it's behavior. I downloaded all the case examples you have provided and was trying to get them working on my system. Before running the whole training routine, I decided to implement the rendering to generate the video as the data was already provided. I downloaded POV ray for windows and placed it in my pyElastica directory. I then ran the files in the order they were mentioned in the readme.txt file, with first the generate_POVray_files.py, render_frames.sh, and lastly make_vid.sh. However I get the following error message in my terminal., 
[image]

and the following files are generated in the images folder.
[image]
Lastly when I run the make_vid.sh file, I see no video being generated.

I am using Anaconda with Pycharm, and when I tried to install POV ray in my pyElastica environment using , https://anaconda.org/conda-forge/povray I encountered the error message, 
PackagesNotFoundError: The following packages are not available from current channels:

  - povray

Current channels:

  - https://conda.anaconda.org/conda-forge/win-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/win-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/win-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://repo.anaconda.com/pkgs/msys2/win-64
  - https://repo.anaconda.com/pkgs/msys2/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

Using pip install also didn't resolve the issue, and I got, ERROR: Could not find a version that satisfies the requirement povray (from versions: none)
ERROR: No matching distribution found for povray

So I just proceeded with downloading and installing POV ray in my pyElastica folder.. I do see the .pov and .inc files in the Images folder, but don't get where the bug is

armantekinalp, 2023-08-08:
Hi @AkashVardhan7 

Here I am copying answer of @Rahul-JOON  (of our contributors):

I actually ran into a similar error while previously correcting the continuum snake example to render video using ffmpeg; Firstly the povray should be installed via the native povray portal [here](https://www.povray.org/download/), next the code doesn't recognise the povray module because the povray module that is installed have the application/executable files named as pvengine, pvengine64 etc present at the C:\Program Files\POV-Ray\v3.7\bin folder.

I fixed this problem by renaming the executable file as 'povray'.

I suggest either doing the same or changing the povray commands in the scripts by specifying the exact module name ex: "pvengine ….".

[image]

AkashVardhan7, 2023-08-08:
I tried both, changing the name of the application file in my extracted folder (I extracted all POV RAY files in this folder called POV_RAY in pyElastica directory - C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\POV_RAY\bin) to povray from both pvengine and pvengine64 and running the code, also changing the script to have pvengine and pvengine64 one after the other which is what I am attaching here, but my issue persists with the code not being able to find the executable
[image]

[image]
 
[image]

AkashVardhan7, 2023-08-08:
Here's the same issue when running the script with pvengine executable
[image]
[image]

Rahul-JOON, 2023-08-08:
Hi @AkashVardhan7, please make sure that the modified executable application file has been added to the "path" environmental variable(see [here](https://www.computerhope.com/issues/ch000549.htm)); in other words, passing 'povray'/'pvengine'(the file name in your bin folder) in cmd should open a povray window as below:

[image]

AkashVardhan7, 2023-08-08:
Rahul thanks for your prompt reply, I have a separate environmental variable for povray and added it's path as well, 
[image]
After that I can open povray window from my cmd prompt,
[image]
But when I run the script , it still doesn;t recognize pvengine/pvengine64

Rahul-JOON, 2023-08-08:
if you have just now added the path, you might need to restart the computer for the changes to take effect, and to run 'pvengine' you don't actually need to 'cd' to the destination folder, you can pass pvengine at any location in the system and it should open povray window.

AkashVardhan7, 2023-08-08:
Rahul, I restarted the computer and re ran the codes, but the issue is still persisting. Also running, pvengine64/pvengine from the home folder in the command prompt returns this error message
C:\Users\avardhan7>pvengine64
'pvengine64' is not recognized as an internal or external command,
operable program or batch file.

I have to run it from the folder where it's installed, which is provided in the previous post. I didn't install povray inside the ProgramFiles rather inside the folder POV_RAY created in my pyElastica directory.. C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\POV_RAY

Rahul-JOON, 2023-08-08:
Hi @AkashVardhan7, I just saw your message above, you have added the path the wrong way; you actually have to add the path inside the path variable by clicking edit and adding a new path then the scripts inside the path provided will be available as commands to the cmd everywhere. I think that's the same reason why you were having trouble with the ffmpeg installation also. Follow the below steps:

1. Copy the destination path
2. open environmental variables
3. click on the path variable
4. click on edit
5. then click on new
6. then paste the path there and apply changes
7. restart the computer
8. verify the installation by running the command in the root directory in cmd.

[image]

AkashVardhan7, 2023-08-08:
Thanks, this fixed it. I can now get POV ray to generate the .png files for me, I have to manually kill the POV -ray window each time an image is generated though, is that how it's supposed to be?

Rahul-JOON, 2023-08-08:
@AkashVardhan7, as i mentioned in #276 's bonus note:
1. open povray
2. go to render options on the top navbar
3. go to on completion
4. check exit povray for windows

This will kill povray instance after every render.
Also, I think now you can even use the continuum snake example properly; just add ffmpeg path the same way.

AkashVardhan7, 2023-08-08:
Thanks, it automatically terminates after each render, I think so too  that I will be able to run continuum snake example now as I also added the ffmpeg path to my Path variable and was able to open it from the root directory in the cmd prompt.. I appreciate your help. :) thanks a lot.

AkashVardhan7, 2023-08-08:
Thevideo for the RL case 3 was generated. I ran continuumsnakeVisualization with POV RAY and encounter the following error,

File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\continuum_snake_visualization\continuum_snake_render.py", line 165, in <module>
    for message in p.imap_unordered(func, batch):
  File "C:\Users\avardhan7\.conda\envs\pyElastica\lib\multiprocessing\pool.py", line 873, in next
    raise value
FileNotFoundError: [WinError 2] The system cannot find the file specified

When I look at the code, I see the following comment statement,
[image]

Any suggestions? The .pov files have been created under the frames folder for both diag and top snakes...

Rahul-JOON, 2023-08-09:
Hi @AkashVardhan7, this error states that it cannot find the file specified while rendering which is the povray module. As mentioned at the top, the native scripts recognize 'povray' files only not 'pvengine'/'pvengine64'.
So just rename the executable application files as 'povray' and the code will be fine.

AkashVardhan7, 2023-08-09:
I copied pvengine and named it to povray and moved the pvengine,pvengine64 out of the bin folder..when I ran the code now I see the following
[image]
Basically POV Ray opens but can't read beyond the first frame, and the rendering process is frozen why is this?

Rahul-JOON, 2023-08-09:
Why did you move the newly named povray file? The file needs to be at the same location as you provided in the path in environmental variables.
You need to check exit povray for Windows after the render option to keep the code running automatically. I think you did this for pvengine64 and not pvengine. Kindly try to implement the needful.

AkashVardhan7, 2023-08-09:
I have the file which I copied from pvengine and then renamed to povray in the bin folder which is added to the PATH variable under environment variables, I removed the other executables pvengine and pvengine64 and placed them
[image]
 outside the bin folder..The error message which I was getting from the povray window as you can see in the previous post was it was unable to open default.inc which is placed in the same folder with the rest of the code.
[image]
[image]
[image]

Rahul-JOON, 2023-08-09:
Please don't move files without a reason; they might be necessary for processing.
The error is still unclear as the code line: (#include "../default.inc") is just fine (I have tested it). Run the example scripts in order.

If the issue still persists; I suggest reinstalling povray and making only the necessary edits that are 1.) renaming pvengine to povray 2.) adding the location as a path in env variables

closed 2023-08-17 by armantekinalp